robotology/mesh-support #1115: R1 microphones not working

randaz81, 2021-04-01:
Only 2 of the 6 microphones seems to properly work. The other 4 microphones are only recording silence.
The problem seems to be an electrical issue since the same software has been test on R1-01 (in Erzelli) and it works.

violadelbono, 2021-04-06:
We will check this together with https://github.com/robotology/icub-tech-support/issues/1114

randaz81, 2021-04-07:
[image]
4 of 6 microphones are now visible on audacity. The internal mics seems to be still missing.
A comparison of the picture with CER02 (currently at Erzelli) will be done to confirm this hypothesis

violadelbono, 2021-04-07:
> 4 of 6 microphones are now visible on audacity.

The left microphone cable was found damaged and fixed

closed 2021-06-23 by github-actions[bot]