robotology/mesh-support #0: yarprobotinterface complains about firmware version and does not start

EliseiFrederic, 2021-04-08:
## Brief description of the request/failure
yarprobotinterface does not start: seems related to a timeout on sharedcan[0]

## Detailed context
no update, nor make, nor repository chane...
BUT continuous use of the robot (always powered on/warm)

## Accompanying material

[yarprobotinterface_log.txt](https://github.com/robotology/icub-tech-support/files/6278607/yarprobotinterface_log.txt)

EliseiFrederic, 2021-04-08:
I also scanned the esw2can buses:
[canloader_report.txt](https://github.com/robotology/icub-tech-support/files/6278627/canloader_report.txt)

EliseiFrederic, 2021-04-08:
[image]

EliseiFrederic, 2021-04-08:
Strange that this issue is numbered 0 ???

pattacini, 2021-04-08:
>Strange that this issue is numbered 0 ???

Definitely, I need to purge it immediately @EliseiFrederic 
Sorry!
Once done, please file a new one.