robotology/mesh-support #1117: Placeholder for Frederic's issue

pattacini, 2021-04-08:
Sorry @EliseiFrederic but I had to react immediately and purge the malformed issue number `0`.
Can you please fill in the details hereinafter again?
Sorry for the inconvenience.

## Brief description of the request/failure

## Detailed context

## Accompanying material

EliseiFrederic, 2021-04-08:
yarprobotinterface refuses to start. Seems related to can [0] (face):

...
[ERROR]sharedcan [0] thread:0 msg:91 joint:2 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version
...
[ERROR]sharedcan [0] thread:0 msg:91 joint:3 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
[ERROR]sharedcan [0] thread:0 msg:91 joint:4 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version 
...
[ERROR]sharedcan [0] thread:0 msg:91 joint:5 timed out
[ERROR]getFirmwareVersion: message timed out
[ERROR]Unable to read firmware version
...
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]  It has been detected that your control boards are not using the same
[ERROR]  CAN protocol used by iCubinterface. iCubInterface cannot continue.
[ERROR]  Please update your system (iCubInterface and/or your control board firmware.
[ERROR]  For further information please visit: http://wiki.icub.org/wiki/Firmware
[ERROR]
[ERROR]###################################################################################
[ERROR]###################################################################################
[ERROR]
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
[yarprobotinterface_log.txt](https://github.com/robotology/icub-tech-support/files/6281620/yarprobotinterface_log.txt)

EliseiFrederic, 2021-04-08:
I also inspected the "replies" on the esw can bus
[canloader_report.txt](https://github.com/robotology/icub-tech-support/files/6281630/canloader_report.txt)
:

EliseiFrederic, 2021-04-08:
Here is what was detected on can bus 0:
[image]

EliseiFrederic, 2021-04-08:
Indeed, removing the 3 faces references in icub_all.xml for our robot
has allowed to run yarprobotinterface.

EliseiFrederic, 2021-04-08:
So, we opened the head for a (random) visual inspection for a wire/connector problem. We didn't see any... but since we closed the head again, we couldn't run yarprobotinterface (other errors for other canbus).
Should we search for a specific connector or wire ? Or do you suspect another error ?

EliseiFrederic, 2021-04-08:
### CONTEXT
The head has been opened several time because of the motherboard battery problem, and the motherboard is powered on continuously 5 days/night per week until we can solder a new battery (we aimed at finishing the current experiment on April the 20th)

violadelbono, 2021-04-09:
Hi @EliseiFrederic, 

Looking at the boards FW version of your robot (v2 -talking head) it seems that the versions are correct. You can find them here:

https://icub-tech-iit.github.io/documentation/icub_firmware/associated-firmware/associated-firmware/#icub-v2-talking-head

I am only confused by board 15 (which in my documentation should be 9) where the protocol is not detected. The whole problem could be that the board doesn't detect the protocol.

- Did you do any maintenance on that board?
- Do you know why the ID is 15 instead of 9?

Anyway I think you can try to flash the fw again on that board and see if the situation changes.

maggia80, 2021-04-09:
yes, I think that the board with ID=15 got an error in the flash area where this ID is saved. And its ID changed from 9 to 15. It could be a failure in the FLASH. Since we are sending you a spare part for the eyelids, we may add an MC4 board.

EliseiFrederic, 2021-04-09:
Thanks @violadelbono & @maggia80 
No maintenance on any board, nor firmware flashing for ages on our side (last flashing was done remotely by @randaz81 & staff). If that helps,I will post of what the canloader scanned for ID9 and ID15.

Should we try to flash ? If yes, can you point me to the procedure ?
Should we disconnect/disactivate the board to continue our experiments (that don't use skin) ? If yes, where can we find it ?

Thanks again

EliseiFrederic, 2021-04-09:
[image]
With the canloader menu, I was proposed ID 0 to 9. Not ID15

maggia80, 2021-04-09:
@EliseiFrederic the problem is that if you try to change the ID of a board with ID=15 it may cause all the board to change their ID to 9, since 15 is the broadcast ID. I'm checking if we have a spare card that we can send you for free.

EliseiFrederic, 2021-04-09:
Would be fine, thanks !

EliseiFrederic, 2021-04-14:
Dear all,

I see the expedition is imminent. Good news :-)

We try to prepare for the repair (never did one such). Were will we find the MC4 ? What will we have to dismantle ?
I tried to follow the links for the v2.3 in [iCub Tech Docs](https://icub-tech-iit.github.io/documentation/icub_wiring/icub2_x/)  but they point to nowhere

Thanks in advance

violadelbono, 2021-04-15:
Hi @EliseiFrederic,

Sorry for the inconvenience, we will fix the website soon.

This is the repository where you can download the electrical schematics: [https://github.com/icub-tech-iit/electronics-wiring-public/tree/master/icub2/icub-2.3.0-talkinghead/pdf](https://github.com/icub-tech-iit/electronics-wiring-public/tree/master/icub2/icub-2.3.0-talkinghead/pdf)

In particular, in iCub 2\_2.3.0\_XXXX\_Logic.pdf you will find the schematic of the head (last page). The MC4 board will come already programmed and with the correct ID 9. What you need to do is to first disconnect all the cables attached to it and replace the old one 0B9 with the new one. The schematic will help you to remember the correct position of every cable attached to the board, but I also suggest to take a preliminary picture before disconnecting, since sometimes it is difficult to recognize all the cables.

[image]

In the Motor&Board placement pdf I see that that board is below 0B7 and 0B1 so unfortunately you need to move them by detaching the stand-offs in order to create space to access the 0B9. 

[image]

[image]

EliseiFrederic, 2021-04-15:
Dear all,Thank you for the files & advices. The board has arrived, and I'll try to find some competents people to assist me (or the reverse...) for the repair task !Thanks again,Frédéric, Loriane, and GérardP.S. Thank you also for the pair of eyelids pairs. I'll probably ask you later how to prepare/screw the two parts together without breaking the thin plastic

EliseiFrederic, 2021-04-15:
Dismantling was ok, except for some nylons screws / hex spacers that we will have to source and replace (M2,5 ?). Looks like they don't like heat and years that much!

EliseiFrederic, 2021-04-16:
Good news, with the new board the robot can start and calibrate as before.
Thanks for the advices, support and already programmed board :+1: !

closed 2021-04-16 by EliseiFrederic