robotology/mesh-support #1118: RX CRC error detected on left leg

S-Dafarra, 2021-04-09:
## Brief description of the request/failure
After starting the robot, I see the following errors:
```
	125,712621	ERROR		hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.13. Expected: 3685, Received: 3686, Missing: 1, Prev Frame TX at 88153211 us, This Frame TX at 88159211 us
	125,722977	ERROR		hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.13. Expected: 3690, Received: 3691, Missing: 1, Prev Frame TX at 88168211 us, This Frame TX at 88174211 us
	125,794990	ERROR		 from BOARD 10.0.1.6 (left_leg-eb6-j0_1), src LOCAL, adr 0, time 88s 230m 290u: (code 0x06000002, par16 0x0001 par64 0x3000000000000002) -> ETH monitor: detected RX CRC error in port specified by par16 (0 = ETH input (P2/P13/J4) , 1 = ETH output (P3/P12/J5) , 2 = internal).  Application state is in most significant nibble of par64: 0 -> N/A, 1 -> idle, 3 -> running. Number of errors in par64&0xffffffff + .
	127,235042	ERROR		hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 3576, Received: 3577, Missing: 1, Prev Frame TX at 89659573 us, This Frame TX at 89669573 us
	127,534633	ERROR		hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 3635, Received: 3636, Missing: 1, Prev Frame TX at 89954573 us, This Frame TX at 89964573 us
	127,554988	ERROR		hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 3638, Received: 3639, Missing: 1, Prev Frame TX at 89969573 us, This Frame TX at 89979573 us
```
These errors appear only once at startup, but even if I restart the motors they appear again.

The robot then starts correctly, but CRC errors are usually no good.

## Detailed context

## Accompanying material

[log_icub-head_yarprobotinterface_29801_CRC_errors.txt](https://github.com/robotology/icub-tech-support/files/6286831/log_icub-head_yarprobotinterface_29801_CRC_errors.txt)

S-Dafarra, 2021-04-09:
This is happening also on the right leg. Here a longer version of the same log I took after trying to take some steps:
[log_icub-head_yarprobotinterface_29801_CRC_errorsx2.txt](https://github.com/robotology/icub-tech-support/files/6286965/log_icub-head_yarprobotinterface_29801_CRC_errorsx2.txt)

pattacini, 2021-04-10:
@S-Dafarra this is the first time you see CRC errors happening on iCub 3 or were there already before the maintenance, if you can remember?

S-Dafarra, 2021-04-10:
I think it is new. I did not find it in the log of https://github.com/robotology/icub-tech-support/issues/1103 for example

pattacini, 2021-04-14:
These messages are detected only at startup. We said that it might be that when the robot reaches its starting joint configuration some wires may be a bit squeezed causing the error.

Let's keep an eye on this for the moment.

S-Dafarra, 2021-04-14:
I also double-checked the log of #1103 and the error is not there. So it has to be something recent.

pattacini, 2021-05-26:
@Mick3Lozzo @salvi-mattia in order to check this problem, we'd need to have sketches similar to https://github.com/icub-tech-iit/electronics-wiring/blob/master/icub3/pdf/iCub%203_Motor%26Placement_LeftUpperarm.pdf but for the iCub 3 legs.

This way, @maggia80 could complete the diagrams with the missing info regarding the boards.

salvi-mattia, 2021-05-26:
the linked ones are the only complete schemes of this kind regardin icub3

i've found the drw (cad) files for the two legs and the waist bu they are without lables

@Mick3Lozzo committes them in the old svn repository, so it seems we were working on it but we stopped before having feedback from electornics regarding where and wich label add to the drawing

[image]

salvi-mattia, 2021-05-26:
the existing, complete pdf
[image]

maggia80, 2021-05-26:
I can work on the `.drw` files

salvi-mattia, 2021-05-27:
I have made them available 
in cad-mechanics\projects\icub\icub3

icub3_eltr_leg_right.drw.1
left_upperarm.asm.1
right_upperarm.asm.1

pattacini, 2021-05-27:
Quick update on this.
@maggia80 is optimizing the cabling as we have seen that there are several critical conditions where the motor wires pass nearby the AMO boards.

pattacini, 2021-07-27:
The wiring has been overhauled.
Please, let us know if this has improved the situation.

S-Dafarra, 2021-07-27:
Yesterday we still noticed some of them. Altough, we were running the ``yarprobotinterface`` from the terminal and we did not take a log. Hence, consider this with a grain of salt. We will check better.

pattacini, 2021-09-27:
Pinned this issue as it is here for being kept under radar 📡

pattacini, 2021-11-23:
> Lately, we mainly observed CRC errors on the torso during the first run (after switching on the motors).

For the torso, we have the #1195.
Closing this as agreed.

closed 2021-11-23 by pattacini