robotology/mesh-support #1119: Shoulder cable of the left arm needs to be replaced.

lunagava, 2021-04-14:
### Request/Failure description

One of the shoulder cable of the left arm (iCubGenova02) is worn and it is likely to break soon.  

@violadelbono 

### Detailed context

As you can see in the following pictures, the tendon is a little frayed. 

### Additional context

[image]
[image]

violadelbono, 2021-04-16:
Hi @lunagava 

It seems that only the Teflon cover of the cable is breaking (correct me if I am wrong), so you can go on for a while before the actual cable breaks.

lunagava, 2021-04-16:
Hi @violadelbono, I heard a strange sound coming from the left shoulder, while the robot is moving. I tried to see if the cable was damaged partially or not. If you have verified that is only the covered plastic broken, I agree to wait until the cable will break definitely.

violadelbono, 2021-04-19:
@lunagava maybe you can show me next time you use it :)

violadelbono, 2021-04-20:
Closing for now, you can open a new one when the cable will actually break

closed 2021-04-20 by violadelbono