robotology/mesh-support #1120: Torso and head loose.

marco-monforte, 2021-04-14:
### Request/Failure description

The torso and head of iCubGenova02 seems a bit loose. I guess that some cables need to be tighten. 

@violadelbono 

### Detailed context

I tried to "manually" move the head and it seems not fixed. 
Looking at the motorgui, the head yaw varies from -0.9 to 1.8 deg. 

Similarly, the torso yaw varies from -2 to +2 deg. Probably, some tendons need to be pulled.

violadelbono, 2021-04-16:
Hi @marco-monforte,

### Torso: 
I tightened the cables. Now, when moved manually, the range of the yaw joint is (-1, +1). Let me know if this is acceptable. I don't know if there is much more we can do.
 
### Neck: 

With the help of @fbiggi and @emilianob80 we opened the head and:

- opened the harmonic drive assemble
- added grease into the bearing
- tightened screws of flex plane

Now the backlash is reduced (-0.9, +0.9). The remaining backlash is due to mechanical wear. A more invasive option could be to change the whole harmonic drive system but it requires time and a proper scheduling.

marco-monforte, 2021-04-19:
Thanks @violadelbono !

With @lunagava we'll do some tests asap

lunagava, 2021-04-19:
Thank you very much @violadelbono! 

Given that tomorrow I will be at CRIS, I will tell you if this is acceptable for my application!

lunagava, 2021-04-21:
Yesterday, I verified the torso is fixed enough, while the head still moves a little, but this is due to usury. The current status of the robot is sufficiently good for doing some tests, so @violadelbono, I think you can close this issue. 

I will open a new issue to substitute the neck, as soon as my demo will be ready.

violadelbono, 2021-04-21:
Alright @lunagava, thanks!

closed 2021-04-21 by violadelbono