robotology/mesh-support #1123: R1 SN003 (cer04) - power supply not working

lnobile, 2021-04-21:
### Robot name 🤖

R1 SN003 (cer04)

### Request/Failure description

The power supply of the robot has stopped working and outputs the following error message: "fail 4".

### Detailed context

Please look at the attached photo

maggia80, 2021-04-21:
```
R1 SN003 (cer04)
```

@lnobile cer04 is deprecated. R1 SN003 is now R1SN003

lnobile, 2021-04-21:
> ```
> R1 SN003 (cer04)
> ```
> 
> @lnobile cer04 is deprecated. R1 SN003 is now R1SN003

Edited ;)

violadelbono, 2021-04-21:
IITCODE is 1596

I didn't find anything useful in the [manual](http://www.av.it.pt/Medidas/Data/Manuais%20&%20Tutoriais/33%20-%20Power%20Supply%20CPX400DP/CPX400D%20&%20DP%20Instruction%20Manual.pdf) 

Here similar problem and someone repaired it https://www.eevblog.com/forum/repair/sorensen-xpf-60-20dp/

cc @emilianob80 @maggia80

lnobile, 2021-04-21:
Do you think that is possible to temporarily use the SN000 power supply? SN000 is actually not used and they share the same power supply. However, the connector cable is different and we should also switch the cables to make it work. (I can do this also on my own, just need permission)

randaz81, 2021-04-21:
Since this is an emergency, I also agree that the quick solution is to switch the cables with SN000 power supply. 
Please ask to @emilianob80 or @MrAndrea for help on the power supply settings.

MrAndrea, 2021-04-21:
I didn't find anything useful information to understand FAIL4 error message like @violadelbono, probably there is a worrying hardware problem inside and now another power supply is needed.

emilianob80, 2021-04-21:
> I didn't find anything useful information to understand FAIL4 error message like @violadelbono, probably there is a worrying hardware problem inside and now another power supply is needed.
> 
> Here a video of a similar [FAIL4](https://arshow.info/watch/1aBJ_t9QagI/sorensen-xpf-60-20dp-failed.html)

https://www.eevblog.com/forum/repair/sorensen-xpf-60-20dp/

lnobile, 2021-04-21:
I swapped the power supply with the one of the old SN000 so that now we can work for now

Andreamenini68, 2021-04-22:
@violadelbono  Allego richiesta info per Assistenza Riparazione
[Mail RS per richiesta Assistenza.pdf](https://github.com/robotology/icub-tech-support/files/6356862/Mail.RS.per.richiesta.Assistenza.pdf)

pc @maggia80  @Fabrizio69

Andreamenini68, 2021-04-27:
@violadelbono  Ti allego risposta da parte del Fornitore RS
[Risposta RS per Alimentatore.pdf](https://github.com/robotology/icub-tech-support/files/6383234/Risposta.RS.per.Alimentatore.pdf)

pc @maggia80  @Fabrizio69

maggia80, 2021-04-27:
@Andreamenini68 puoi provare a sentire il produttore? Se fanno riparazioni fuori dalla garanzia

Andreamenini68, 2021-04-28:
@maggia80  certo ho sentito Gardella in quanto se passiamo dalla casa madre le tempistiche sono lunghissime

violadelbono, 2021-05-18:
Ciao @Andreamenini68, ci sono aggiornamenti su questa issue?

Andreamenini68, 2021-05-20:
@violadelbono  Gardella non riesce a fare nulla , abbiamo contattato la casa madre stiamo attendendo risposta, sollecitiamo.

Fabrizio69, 2021-09-15:
@Andreamenini68 ci sono novità in merito a questa issue? Hai ricevuto info?

Andreamenini68, 2021-09-17:
@Fabrizio69  ho sollecitato la Silvia in quanto aveva scritto lei alla casa madre. Ti aggiorno quanto prima

Fabrizio69, 2021-11-24:
@Andreamenini68 ci sono stati aggiornamenti in merito?

cc: @maggia80

Andreamenini68, 2021-11-24:
@Fabrizio69  Ciao Fabrizio, nessun aggiornamento in merito , per sicurezza faccio un passaggio sulla Silvia in quanto ha scritto lei.

silviacriscenzo, 2021-11-29:
Ciao @Fabrizio69 sono riuscita ad ottenere un riscontro dalla casa madre. Dovrebbero riuscire ad occuparsi della riparazione, ma vorrebbero sapere il seriale dell'alimentatore. Io ho verificato sul DDT di RS e non c'è scritto. Potete gentilmente fornirmelo? Grz mille @Andreamenini68

LaMelis, 2021-12-31:
Ciao @Fabrizio69 riusciresti per caso a fornirmi il num di serie dell'alimentatore? Così posso rispondere alla casa madre. Grazie!

cc @maggia80 @Andreamenini68

Andreamenini68, 2021-12-31:
@LaMelis  eccolo Serial number: 504849

Andreamenini68, 2021-12-31:
@LaMelis  se ti manca qualcosa a disposizione

LaMelis, 2021-12-31:
Grazie @Andreamenini68 ho risposto alla casa madre. Vi aggiorno non appena avrò riscontro.

Fabrizio69, 2022-01-11:
Ciao @LaMelis , scusa se non ho risposto alla [tua richiesta](https://github.com/robotology/icub-tech-support/issues/1123#issuecomment-1003315671), ma in dipartimento, non sono riuscito a trovare l'alimentatore. Forse in questo è riuscito @Andreamenini68 che ti ha fornito [la risposta](https://github.com/robotology/icub-tech-support/issues/1123#issuecomment-1003391520).
Scusami ancora....
Grazie a tutti

cc: @maggia80

Andreamenini68, 2022-01-11:
@Fabrizio69 io dell'alimentatore non so nulla !!!!

cc: @maggia80 @LaMelis

emilianob80, 2022-01-12:
@Fabrizio69 @Andreamenini68 l' alimentatore è in laboratorio con relativa scatola. Il numero seriale è 504849

cc: @maggia80 @LaMelis

Fabrizio69, 2022-01-12:
Ciao @emilianob80 potresti per favore consegnarlo a @Miche19 in modo che lo metta tra il materiale da spedire e possa qui indicare peso e dimensione del collo contenente:

1 pezzo cod.1596 - Power supply with digital BUS interfaces, 2x 420W, 0-60V, 0-20A, 210x130x375mm, Portable, Input 110V to 240V 

@Andreamenini68 penso che valga la pena riparare questo alimentatore, quindi se @maggia80  è d'accordo procederei con la riparazione. In caso contrario procederei con lo smaltimento.
Grazie mille a tutti

cc: @LaMelis

emilianob80, 2022-01-12:
> Ciao @emilianob80 potresti per favore consegnarlo a @Miche19 in modo che lo metta tra il materiale da spedire e possa qui indicare peso e dimensione del collo contenente:
>
> 1 pezzo cod.1596 - Power [...]

ok

LaMelis, 2022-01-12:
Ciao a tutti. 
Volevo solo informarvi che la casa madre (Thurlby Thandar Instruments Ltd), l'unica che può procedere con la riparazione dell'alimentatore, non è iscritta al nostro albo fornitori. Pertanto, nel caso in cui si decidesse di continuare, vi chiedo di tenere in considerazione la loro volontà o meno di iscriversi e le tempistiche dell'iscrizione. Grazie! 

@maggia80  @Fabrizio69 @emilianob80 @Andreamenini68 @zapp

Andreamenini68, 2022-01-12:
@Fabrizio69  Ho già avvertito Salvatore per preparare la documentazione e predisporre la spedizione. Il tutto procederà dopo l'apertura della parte aministrativa.
cc: @maggia80 @LaMelis

Miche19, 2022-01-12:
> Ciao @emilianob80 potresti per favore consegnarlo a @Miche19 in modo che lo metta tra il materiale da spedire e possa qui indicare peso e dimensione del collo contenente:
>
> 1 pezzo cod.1596 - Power [...]

Ciao @Andreamenini68 il materiale è pronto per essere ritirato

Si tratta di una scatola di cartone.
Dimensioni: 30x35x50 cm Peso: 8,5 kg

cc: @Fabrizio69 @emilianob80 @LaMelis

closed 2022-01-12 by maggia80