robotology/mesh-support #1124: Faulty Fault Button - iCub3

davidetome, 2021-04-21:
### Robot name 🤖

iCubGenova09

### Request/Failure description

Fault button not working

### Detailed context

I noticed that the robot behaves as the fault button is always released even if physically detached.

### Additional context

cc @pattacini @S-Dafarra @Fabrizio69

pattacini, 2021-04-21:
This is a quite blocking issue, unfortunately.

davidetome, 2021-04-21:
cc @isorrentino

Fabrizio69, 2021-04-21:
Next Friday @emilianob80 can support @violadelbono for this issue

violadelbono, 2021-04-23:
The fault issue was solved and we found the following problems:

- The fault cable on the BAT board was connected to the wrong connector
- The left leg caused the fault to be kept at ground. Probably due to a cover screw that we noticed pinching the cable

The leg cabling was fixed and the connector on the BAT was switched in the correct place. The robot was tested and now the fault button is fully working.

cc @MrAndrea

closed 2021-04-23 by violadelbono

MrAndrea, 2021-04-23:
The Fault wire was connected on the BAT connector J22 which has 12V output instead of the right connector J23, luckily it seems that this high voltage didn't compromise the boards.