robotology/mesh-support #1125: iCubGenova09 fault broken

isorrentino, 2021-04-26:
### Robot name 🤖

iCubGenova09

### Request/Failure description

The fault on iCubGenova09 broke when the robot started. 

[image]
[image]
[image]

### Detailed context

The fault cable was blocked under the cart wheel and as soon as I started the robot through `yarprobotinterface` the fault jack broke and the connector into the robot torso as well.

MrAndrea, 2021-04-26:
I removed the back cover and I saw that the `FED` board was damaged because the jack stereo of the Fault torn the pcb nets.
The white part was broken and I put some drop of Pattex glue to join the two pieces.

[image]

[image]

I prepared a new `FED`  board and tomorrow Davide Gandini will repair the fault cable.

cc @Fabrizio69 @maggia80

isorrentino, 2021-04-26:
Thanks @MrAndrea!

Gandoo, 2021-04-27:
[image]

[image]

[image]

ciao @MrAndrea la patex non ha risolto la rottura, ho messo attak con dello scotch come rinforzo momentaneo, Mattia lo fa costruire e appena possibile si sostituisce. 

[image]

Risaldato jack al cavo di Fault

cc @Fabrizio69 @maggia80 @isorrentino

MrAndrea, 2021-04-27:
Grazie @Gandoo, avevo provato la Pattex ma forse era meglio darci subito l'Attack, comunque il pezzo ristampato andrà sicuramente meglio.

MrAndrea, 2021-04-27:
@salvi-mattia possiamo ristampare il supporto bianco con un diametro maggiore per il foro cerchiato in rosso? può avere il diametro del foro che c'è nella cover nera. Il motivo é che il jack tocca sui bordi del foro e non arriva in fondo al connettore sulla scheda.

[image]

Gandoo, 2021-04-27:
[image]

[image]

ciao, @MrAndrea  come scritto prima il problema del Fault era dovuto al fatto che il jack del pulsante non andava in battuta e relativa sede del connettore femmina sulla scheda. (FORSE DOVUTO AL FATTO CHE IL SUPPORTO SI è SPEZZATO PROPRIO NELLA ZONA DI FISSAGGIO )

@salvi-mattia ho allargato sia il foro sel supporto bianco che il foro della cover nera sulla parte bassa.

@davidetome  ha fatto il test di verifica del funzionamento del Fault, risulta funzionante.

cc @Fabrizio69 @maggia80 @isorrentino

salvi-mattia, 2021-04-27:
avendo recepito le richieste soprastanti

[image]

salvi-mattia, 2021-04-27:
[image]

violadelbono, 2021-05-12:
Ciao @salvi-mattia,

Hai novitá su questa issue? E' arrivato il pezzo?

salvi-mattia, 2021-05-13:
si, e' arrivato da tempo
@Fabrizio69 aveva detto di tenerlo a magazzino come spare, al momento lo ho io.
se lo vogliamo montare ve lo porto

cc @violadelbono

salvi-mattia, 2021-05-13:
here It Is on your desk @violadelbono 

[image]

violadelbono, 2021-05-18:
We decided to store the new piece in the warehouse and to eventually replace it when it will break totally.
Closing

closed 2021-05-18 by violadelbono