robotology/mesh-support #1126: Connecting monitor to icub-head

amaroyo, 2021-04-26:
### Robot name 🤖

iCubWaterloo01

### Request/Failure description

Can't get any signal from the displayport of iCub's head. 

### Detailed context

Hello, 

As I am trying to upgrade iCub's head from Ubunty 18 to 20; I connected a monitor directly to the head via the displayport. Unfortunately, I don't receive any signal. 

I don't have a displayport cable, so I used an adaptor with an hdmi cable, it didn't work. I tried a different adaptor, and a simple vga cable.. but didn't work either. I doubt that's the adaptor/cables but I wanted to double check with you for any suggestions/advice. 

Thanks,
Alex.

violadelbono, 2021-05-03:
Hi @amaroyo,

Could it be that the setting of your monitor are not correct?
Usually, when connecting a monitor, you have to set the mode as HDMI, Displayport, etc..

Let me know,
Viola

amaroyo, 2021-05-04:
Hi @violadelbono, 

No, it wasn't the settings :D Actually I bought a hdmi/display port direct cable and it worked.... It seems iCub doesn't like displayport adapters (as I tried 2 different). 

I guess it's a useful note to add in the wiki! Or change the video output of the head... It's getting difficult to find displayport cables :P 

Feel free to close the issue!
Alex.

violadelbono, 2021-05-04:
Very strange, I will investigate on this. Thanks for the feedback!

closed 2021-05-04 by violadelbono