robotology/mesh-support #1127: iCubGenova09 torso joint 1 AMO not working

isorrentino, 2021-04-27:
### Robot name 🤖

iCubGenova09

### Request/Failure description

After https://github.com/robotology/icub-tech-support/issues/1125 the torso pitch motor is not starting properly. 

### Detailed context

Maybe, the joint 1 reading is not working due to a broken cable.

cc @davi

isorrentino, 2021-04-27:
cc @S-Dafarra

davidetome, 2021-04-27:
The sensor is connected to the `EMS` board `EB5` connector `P8`

cc @pattacini @simeonedussoni

pattacini, 2021-04-27:
We will check it out!

pattacini, 2021-04-29:
@davidetome and @simeonedussoni fixed it up.
We will finalize the work on this today.

davidetome, 2021-04-29:
@gandoo 🏅  repaired the fault by crimping the 2 Molex connectors of the sensor cable

cc @pattacini @simeonedussoni @S-Dafarra @isorrentino

closed 2021-04-29 by davidetome

isorrentino, 2021-04-29:
Thank you all!