robotology/mesh-support #1130: Left hand eb25-j8_11 goes to not-configured status under stress

kouroshD, 2021-05-04:
### Robot name 🤖

icubGenova09

### Request/Failure description

Following the issue in https://github.com/robotology/icub-tech-support/issues/1048, we still face the problem with iCub left hand eb25-j8_11 board. 

### Detailed context

When the robot interacts with the environment or when there is a large error between the robot desired position direct desired and feedback value (i.e., under stress), the board goes to `not-configured` status and as a result we need to restart the `yarprobotinterface`. This happened during teleoperation scenarios with `position-direct` control mode. Later, we will provide further information that can help for debugging. Back in time we tried to reduce the PWM maximum vallue (if I am not it was [`fullscalePWM`](https://github.com/robotology/robots-configuration/blob/74962c40b1335f6ba00b82772db12f576b83a16a/iCubGenova09/hardware/mechanicals/left_arm-eb25-j8_11-mec.xml#L14)), but this is not a viable solution since it significantly degrades the control performance.

CC @DanielePucci  @S-Dafarra

S-Dafarra, 2021-05-04:
This is something that does not happen on the iCubGenova04. It is crucial that the joints do not reach this unrecoverable state.

cc @pattacini @maggia80

pattacini, 2021-05-04:
Looking at the past issue as well, it'd worth replacing the board, probably.

S-Dafarra, 2021-05-04:
@kouroshD does it happens on both arms?

marcoaccame, 2021-05-04:
First suggestion is to post the whole log file. From that we can see if the board resets and hence restarts.
If I am not wrong (pls @ale-git confirm it) at startup the board puts the joints in `not-configured` state.

kouroshD, 2021-05-21:
Here is the result for the left hand, when it goes to not-configured status:

[image]
[image]
[image]
[image]
[data-plotter.zip](https://github.com/robotology/icub-tech-support/files/6523441/data-plotter.zip)

I have added the logged data file and a MATLAB code to plot the data.

@pattacini @marcoaccame @S-Dafarra

kouroshD, 2021-05-21:
I tried with the right hand as well. 

[image]

[image]
[image]

[image]

**It happened after 40 seconds, when start grasping an object.**
Before that, the tracking is not good as well due to https://github.com/robotology/icub-tech-support/issues/1058#issuecomment-845898772 which is only related to right hand

It goes to not-configured status, similar to left hand.

@pattacini @marcoaccame @S-Dafarra @DanielePucci @ale-git  @maggia80

pattacini, 2021-05-21:
Thanks @kouroshD 👍🏻 
I reckon that @marcoaccame aimed to see the log of the `yarplogger` as well in order to inspect some diagnostics. Do you have it?

Also, could you please clarify the following points:
- What's the range of the finger joints of the left hand? Are they scaled somehow? They seem to be moving of just fractions of degrees. The right hand, instead, looks better in this respect.
- What are the units of the x-axis? Seconds? There seems to be a mismatch between the left and right hand in this regard.

kouroshD, 2021-05-21:
> What's the range of the finger joints of the left hand? Are they scaled somehow? They seem to be moving of just fractions of degrees. The right hand, instead, looks better in this respect.

My bad, I did not transformed them to degree. They were in radians. I have updated the figures.

> What are the units of the x-axis? Seconds? There seems to be a time mismatch between the left and right hand in this regard.

Again, it was samples in that case. I have updated the figures. They are in seconds now.

kouroshD, 2021-05-21:
@pattacini  This is yarprobotinterface logger info:

[log_icub-head_yarprobotinterface_16718.txt](https://github.com/robotology/icub-tech-support/files/6523963/log_icub-head_yarprobotinterface_16718.txt)

maggia80, 2021-05-24:
```
22090,471369 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.29 (right_arm-eb29-j12_15) has shown after being lost for 2.13621 sec 
22090,471378 <ERROR> hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.7. Expected: 550729, Received: 551157, Missing: 428, Prev Frame TX at 2829880356 us, This Frame TX at 2832025356 us 
22090,471387 <ERROR> hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.26. Expected: 556202, Received: 556631, Missing: 429, Prev Frame TX at 2830254767 us, This Frame TX at 2832399767 us 
22090,471396 <ERROR> hostTransceiver()::parse() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 550725, Received: 551153, Missing: 428, Prev Frame TX at 2829867983 us, This Frame TX at 2832012983 us 
22090,471406 <ERROR>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 2832s 62m 367u: (code 0x00000030, par16 0x00d5 par64 0x00000000000422d8) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471417 <ERROR> theNVmanager::Impl::ask() had a timeout for BOARD left_arm-eb26-j12_15 IP 10.0.1.26 and nv ID32 = 0x0100010b -> IND = 1, TAG = eoprot_tag_mc_joint_status_target 
22090,471426 <WARNING> EthResource::getRemoteValue() cannot have a reply from BOARD left_arm-eb26-j12_15 with IP 10.0.1.26 at attempt # 1 w/ timeout of 0.2 seconds 
22090,471435 <ERROR>  from BOARD 10.0.1.4 (right_arm-eb4-j2_3), src LOCAL, adr 0, time 2832s 42m 736u: (code 0x00000030, par16 0x00d4 par64 0x00000000000420f6) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471447 <ERROR>   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD left_arm-eb26-j12_15 with IP 10.0.1.26  even after 0.200191 seconds: CANNOT PROCEED ANY FURTHER 
22090,471457 <ERROR>  from BOARD 10.0.1.8 (left_leg-eb8-j4_5), src LOCAL, adr 0, time 2832s 15m 394u: (code 0x00000030, par16 0x00d4 par64 0x0000000000042117) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471467 <ERROR> embObjMotionControl::getRefDutyCycleRaw() could not read openloop reference for  BOARD left_arm-eb26-j12_15 (IP 10.0.1.26)  joint  1 
22090,471476 <ERROR>  from BOARD 10.0.1.27 (right_arm-eb27-j4_7), src LOCAL, adr 0, time 2832s 419m 855u: (code 0x00000030, par16 0x00d4 par64 0x0000000000040fa7) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471487 <ERROR>  from BOARD 10.0.1.12 (right_leg-eb12-j4_5), src LOCAL, adr 0, time 2832s 31m 416u: (code 0x00000030, par16 0x00d4 par64 0x0000000000042df5) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471498 <ERROR>  from BOARD 10.0.1.11 (right_leg-eb11-j2_3), src LOCAL, adr 0, time 2832s 42m 709u: (code 0x00000030, par16 0x00d4 par64 0x0000000000042db7) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471508 <ERROR>  from BOARD 10.0.1.3 (right_arm-eb3-j0_1), src LOCAL, adr 0, time 2832s 38m 629u: (code 0x00000030, par16 0x00d4 par64 0x000000000004210a) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,471519 <ERROR>  from BOARD 10.0.1.10 (right_leg-eb10-j0_1), src LOCAL, adr 0, time 2832s 4m 442u: (code 0x00000030, par16 0x00d4 par64 0x0000000000042e13) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,481279 <ERROR>  from BOARD 10.0.1.7 (left_leg-eb7-j2_3), src LOCAL, adr 0, time 2832s 28m 769u: (code 0x00000030, par16 0x00d4 par64 0x000000000004210d) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,481305 <ERROR>  from BOARD 10.0.1.6 (left_leg-eb6-j0_1), src LOCAL, adr 0, time 2832s 4m 172u: (code 0x00000030, par16 0x00d4 par64 0x0000000000042117) -> SYS: the board has detected a hole in rx ropframe sequence number. In par64 there is the expected number. The received is obtained by adding par64 with par16 treated as a int16_t + . 
22090,481321 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.7 (left_leg-eb7-j2_3) has shown after being lost for 2.14628 sec 
22090,481335 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.8 (left_leg-eb8-j4_5) has shown after being lost for 2.1463 sec 
22090,481347 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.26 (left_arm-eb26-j12_15) has shown after being lost for 2.14633 sec 
[... cut ...]
22090,551910 <WARNING> EthResource::getRemoteValue() cannot have a reply from BOARD left_arm-eb26-j12_15 with IP 10.0.1.26 at attempt # 1 w/ timeout of 0.2 seconds 
22090,551924 <ERROR>   FATAL: EthResource::getRemoteValue() DID NOT have replies from BOARD left_arm-eb26-j12_15 with IP 10.0.1.26  even after 0.200177 seconds: CANNOT PROCEED ANY FURTHER 
22090,551934 <ERROR> Failure of askRemoteValue() inside embObjMotionControl::checkMotionDoneRaw(j= 2 ) for  BOARD left_arm-eb26-j12_15 (IP 10.0.1.26)  
22090,561983 <DEBUG> eth::EthMonitorPresence: BOARD 10.0.1.25 (left_arm-eb25-j8_11) has been silent for 0.020292 sec (its timeout is 0.02 sec) 
```

maggia80, 2021-05-24:
it seems to me somehow a congestion in the ETH bus, causing an avalanche of errors and misbehaviors

marcoaccame, 2021-05-24:
I can see a lot of CRC errors detected in the ETH communication. In thus cases, the UDP frames are not delivered and there are a lot of error messages such as: `detected an ERROR in sequence number`, `DID NOT have replies from BOARD`, `eth::EthMonitorPresence: BOARD 10.0.1.7 (left_leg-eb7-j2_3) has shown after being lost for 2.14628 sec` etc,

We must remove the CRC errors from the robot otherwise we cannot operate on it.

I am going now to map where in the robot these CRC errors happen.

It would be important to verify if the CRC errors are triggered by the grasping action or  if they are independent of that. Maybe a joint moves and sends a current leak on the robot which triggers CRC errors in ETH.

marcoaccame, 2021-05-24:
Here is what I found that surely does problems: ETH CRC errors, ETH link up and down, one board restarts. 

The CRC errors happen in boards:
- `10.0.1.10` (right_leg-eb10-j0_1), `10.0.1.11` (right_leg-eb11-j2_3) both in out port;
- `10.0.1.12` (right_leg-eb12-j4_5) in input port; (it detects the 10.0.1.11 out port)
- `10.0.1.6` (left_leg-eb6-j0_1) in output port

Also found that the ETH link goes down and then up again in following boards:
- link down in `10.0.1.24` (left_arm-eb24-j4_7), out port @ YARP time 22083,005638
  there is now a hole in the network for about 2 seconds in when no board in daisy chain after `10.0.1.24` can communicate.
- link up in `10.0.1.24` (left_arm-eb24-j4_7), out port @ YARP time 22085,105084
- link up in `10.0.1.26` (left_arm-eb26-j12_15), input port @ YARP time 22085,134987. It is the board after the `.24`.

The same pattern (`.24` link down and then up two seconds later) happens again at YARP time  22088,413366.

I can also see that board `10.0.1.25` (left_arm-eb25-j8_11) restarts at YARP time 22097,490514.

marcoaccame, 2021-05-24:
@kouroshD and @pattacini: how do we relate the time inside the log of yarplogger w/ the time reported in https://github.com/robotology/icub-tech-support/issues/1130#issuecomment-846033095?

marcoaccame, 2021-05-24:
here is my hypothesis:

- at a certain time around YARP time 22083,005638 the board `10.0.1.25` (left_arm-eb25-j8_11) has problems and ... puts its ETH link down. maybe a reset.
- the board `10.0.1.24` which is before in the ETH daisy chain detects at time 22083,005638 that its output link is down (it is the link towards the `10.0.1.25` having problems)
- two seconds later at time 22085,105084 the boards detect that the same out link is up again. That means that the  `10.0.1.25`  is now OK in terms of ETH commnication.
- at the same time, the board `10.0.1.26 `tells that its input link is also up again (the is the output link of board .25).
- at time 22097,490514 the board `10.0.1.25` transmits again and tells that ... it has just started.

So, the board `10.0.1.25` simply restarts. And after that it is certain that it stays in state `NOT CONFIGURED`.

Question now is: why does it reset and restart? I haven't enabled any restart mechanism in the FW, so if it is related to the hand grasping I would look at at some electrical noise effect or short-circuit due to uncovered cable.

pattacini, 2021-05-24:
> @kouroshD and @pattacini: how do we relate the time inside the log of yarplogger w/ the time reported in [#1130 (comment)](https://github.com/robotology/icub-tech-support/issues/1130#issuecomment-846033095)?

it depends on how @kouroshD did his acquisitions. If the system is time synched it's likely that the clock is correctly shared and the plots can refer to an absolute time baseline, although above the time origin has been clearly made equal to `0 s` for the sake of the readability, I suppose.

kouroshD, 2021-05-24:
@pattacini  @marcoaccame I used https://github.com/robotology/walking-teleoperation/blob/4be02a0abb3881ecb1a6015ee6ae6cb035b03b35/modules/HapticGlove_module/src/HapticGloveModule.cpp#L1329 yarp time to log the data. If you open the logged file with the matlab codes, you can easily find the yarp timing. Otherwise, I can tell you.

kouroshD, 2021-05-24:
For the left hand, the zero is at `1621609260.44006` and for the right hand it is the zero is `1621610324.83029`.

marcoaccame, 2021-05-24:
Ok, thanks. 

I wanted to reference the time `22083,005638` as expressed inside the log file because that is the moment in which the board restarts and then goes in `not-configured` state. But this number does not relate easily to the YARP time `1621609260.44006` you wrote, so probably the `yarplogger` uses a different timeline.

But never mind. From [here](https://github.com/robotology/icub-tech-support/issues/1130#issuecomment-846033095) we have related the grasping activity to the moment in which the board goes in `not-configured` state. And from [here](https://github.com/robotology/icub-tech-support/issues/1130#issuecomment-847023042) we found out the cause of going into `not-configured`: the board resets.

To solve the issue we need to understand why the board restarts when the hand grasps.

kouroshD, 2021-06-04:
Two videos showing how replicate the error. I suggest to try at the moment of the left hand, since there are some other problem with the right hand (refer to https://github.com/robotology/icub-tech-support/issues/1058):

https://user-images.githubusercontent.com/17707730/120784377-4d3cdf80-c52c-11eb-93d3-c79cfec18eda.mp4

pattacini, 2021-06-04:
Very nice visual explanation of the experiment @kouroshD! Thanks a lot!

We are finalizing our report on the AMO problems today (cc @marcoaccame) so that we will be dealing with this in the first days of the next week.

cc @davidetome

pattacini, 2021-06-08:
We can look at this starting tomorrow.

cc @davidetome

davidetome, 2021-06-09:
Today w/ @emilianob80 we unmounted the left forearm cover and we noticed that the boards went down.
After rebooting we tested , w/ the kind help of @kouroshD, the fingers of both hands trying to reproduce the failure both w/ the `yarpmotorgui` and teleoperation many times but the failure seems disappeared also stressing a lot the motors.

Probably the cover was stressing some wires.
Keep this issue open to see if the failure comes back.

cc @pattacini @S-Dafarra @maggia80 @Fabrizio69

S-Dafarra, 2021-06-30:
This happened also during the video shot the 23rd of June.

DanielePucci, 2021-08-30:
Probably we need further investigation on this issue?

pattacini, 2021-08-30:
Hi @DanielePucci
I told @S-Dafarra that we're awaiting feedback from you after @emilianob80 re-wired the forearms.
Before doing that, @emilianob80 spotted several wires in bad conditions, actually.

S-Dafarra, 2021-11-02:
Keeping it up for the moment.

Uboldi80, 2021-11-22:
I think this problem could be solved by the last intervention on the left hand.
@let me know @S-Dafarra if the problem persist.

pattacini, 2021-11-24:
Any feedback on this longstanding issue, @kouroshD @S-Dafarra?

S-Dafarra, 2021-11-24:
Given the amount of time spent lately to fix the arms, we avoided undergoing much stress on the hands in this period filled with demos, to be honest.

pattacini, 2021-11-24:
No problem! Keep us posted 👍🏻

Uboldi80, 2022-02-08:
following the interventions carried out and the time dedicated to the robot, this problem has arisen more

pattacini, 2022-03-07:
I think we can close this issue.
Feel free to open it up again if needed.

closed 2022-03-07 by pattacini