robotology/mesh-support #1131: R1 SN003 new back cover doen't properly close

lnobile, 2021-05-04:
### Robot name 🤖

R1 SN003

### Request/Failure description

The new R1 SN003 new back cover doen't properly close since it has an interference with the nvidia xavier board mounted on the robot.

### Detailed context

@fbiggi already cutted out a part of the cover to make it fit so this is no more an issue but just a remainder to update the drawings in the future.
The part cutted (see photo) was in contact with the fan housing of the Nvidia xavier board.
[image]

fbiggi, 2021-05-04:
the cover actually strained on the rear fan
I preferred to cut and shape
It seems that closing is much easier

cc @fiorisi @maggia80 @Fabrizio69

fiorisi, 2021-05-04:
@lnobile having a look at the CAD model it seems that there shouldn't be any interference. I think that we should understand which is the difference between the CAD model and the real prototype.
[image]
[image]
[image]

lnobile, 2021-05-04:
It seems to me that the upper ribbon of the cover is at the same height of the nvidia xavier's fan.
I don't know if the ribbon is lower or the fan higher, but this seems the problem.
[image]

lnobile, 2021-05-04:
The board in CAD seems below the white plastic plate under the shoulders motors, while on the real robot the upper part of the board is at the same height of the white plate

lnobile, 2021-05-04:
Ok I got it, the board is mounted upside down (look at the connectors)

maggia80, 2021-05-04:
yes! Indeed the board is mounted upside down. If there aren't specific reasons we could put it in the "right" position.

lnobile, 2021-05-04:
It doesn't seem to me, connector cables have all sufficient length.
Maybe @randaz81 or @emilianob80 could know any possible reason

randaz81, 2021-05-04:
> Maybe @randaz81 or @emilianob80 could know any possible reason
No idea. Are you sure that every connector (usb/network/hdmi) have enough space for insertion? Eventually we might need to buy cables with angled connectors.

fbiggi, 2021-05-05:
la casa di R1é in lab
@MrAndrea @marcellosavoldi non posso fare le prove di imballaggio con la cover nuova
Quando potete darmelo rimontato?

MrAndrea, 2021-05-05:
@fbiggi oggi dovremmo riuscire a richiudere la testa di R1. @violadelbono sta preparando i cavetti per gli speaker e per l'amplificatore audio, e non appena pronti procederemo al cablaggio e al test e a rimontare le cover.

cc @marcellosavoldi @maggia80

MrAndrea, 2021-05-05:
Nel pomeriggio con @violadelbono ho collegato l'amplificatore ma purtroppo gli speaker gracchiano molto e di fatto l'audio si sente poco e male. Provando a cambiare un po' i collegamenti per trovare la causa del problema il risultato è stato sempre lo stesso, e sembra indipendente dal volume che regola dal pc con l'alsamixer.
Infine anche con @maggia80 ho riprovato i test ma visti gli stessi risultati abbiamo deciso di ripristinare il vecchio collegamento e richiudere la testa.

@fbiggi il robot è rimontato e puoi provare la cassa.

fbiggi, 2021-05-06:
ATTENZIONE !!!
Con @lnobile abbiamo provato a mettere il robot R1 con la back cover nuova per XAVIER
IL robot tocca e scarica il peso sulla cover nuova ...e la pedana non si riesce a togliere.
Senza cover la XAVIER tocca e potrebbe danneggiarla.
Abbiamo deciso dopo il lavoro di @lnobile  con il team di smontare la XAVIER 
Datemi info e conferme

@maggia80 @randaz81 @Fabrizio69 @emilianob80 @violadelbono @fiorisi