robotology/mesh-support #1132: R1 SN003 new back cover hides fault button inside the robot

lnobile, 2021-05-04:
### Robot name 🤖

R1 SN003

### Request/Failure description

R1 SN003 new back cover hides fault button inside the robot

### Detailed context

As you can see from the photo the falt button is inside the robot once the cover is closed.
Is it possible to connect an external fault button or make the existing one accessible?
[image]

maggia80, 2021-05-04:
@lnobile yes we added the possibility to use an external fault. We can do a remote fault. Currently the connector is closed with a jumper inside the robot.

[image]

maggia80, 2021-05-04:
cc @emilianob80 @Fabrizio69

emilianob80, 2021-05-04:
@lnobile  tomorrow morning I'll wire an iCub-style remote fault button. The length will be 3 meters

cc: @maggia80 @Fabrizio69

lnobile, 2021-05-04:
> @lnobile tomorrow morning I'll wire an iCub-style remote fault button. The length will be 3 meters
> 
> cc: @maggia80 @Fabrizio69 @violadelbono

Great! But given the fact that we will need the robot to navigate autonomously around the museum maybe it's better a shorter cable so that it can be easily stuffed inside the cover and the external fault button attacched/hanged on the robot

maggia80, 2021-05-04:
@lnobile you can plug the jumper when you don't want to use the fault button.

emilianob80, 2021-05-05:
wired external fault
 
[image]

cc: @maggia80 @lnobile @randaz81 @violadelbono

lnobile, 2021-05-05:
great! thank you!

violadelbono, 2021-05-05:
Robot launched and faulted multiple times ✔️ everything works

closed 2021-05-05 by violadelbono

emilianob80, 2021-05-05:
@lnobile When you want to connect the external fault, disconnect the jumper in the photo. The connector has a retainer, so first press the tab and then carefully pull

[image]

[image]

cc: @violadelbono