robotology/mesh-support #1135: Hardware fault of the left shoulder when oscillating

lindalastrico, 2021-05-14:
### Robot name 🤖

iCubGenova07

### Request/Failure description

When the left shoulder of the robot is raised, there is a fault of the shoulder joints if the arm is oscillating back and forward. An example of the arm position when the fault occurred is: pos (**-27.0 78.0 -30.0** 33.0 -79.0 0.0 -4.0 49.0 7.0 0.0 39.0 49.0 87.0 67.0 130.0 200.0). 
We observed this problem several times either when the robot was moving on its own or when we tried to shook it slightly when the shoulder joints were in that position.
We tested if the same would happen on the right arm, but we did not experience any similar issue.

### Detailed context

Yarp logger output:

4666	487,948161	ERROR	 from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 6497s 787m 49u: (code 0x02000008, par16 0x0001 par64 0x0000000000000009) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

Uboldi80, 2021-05-18:
Ciao @lindalastrico 

In funzione dei nostri impegni schedulati, non potremmo passare da voi prima di Martedì 25.
Può andare bene?

lindalastrico, 2021-05-18:
Sì va bene, grazie! Vi aspettiamo

Uboldi80, 2021-05-25:
We fix the problem on to  **AEA connector LOS1**.
the problem was fixed by crimp in a black wire.
🚶 🚶 🚶 🚶 
And then, **another problem fixed on the neck routing**.
Probably neck cable  not tensioned enough and was exit to your pulley.

closed 2021-05-25 by Uboldi80