robotology/mesh-support #1136: assembly both arms icubGenova SN.031

Uboldi80, 2021-05-21:
change the hardware configuration of the robot to reassembly both arms after experimental fly.

Uboldi80, 2021-05-21:
right arm is assembled and tested
some problem in the joint 8-9

Uboldi80, 2021-05-21:
left arm is assembled but we have few problems hardware:
 
[image]

some parts are missing, like a  upper arm cover and a finger tip

kouroshD, 2021-05-28:
After the assembly has been performed, we have the following problem with the right arm and hand:

the right thumb oppose encoder feedback is high, so it is stays in HW fault

[image]

We face a couple of times right elbow was not configuring.
This is the image for the right prosup, however, later we fault the robot and unfault the robot, and restrarted the robot and it started working. However, for the sake of reporting, I am mentioning it here:

[image]

An issue was closed about this recently, in https://github.com/robotology/icub-tech-support/issues/1111

I can say also the right wrist prosup is not working as expected:
In this case, we were moving the right prosup to the whole possible range of motion using yarp motor gui, while in reality it was not moving more than 20-30 degrees. You can see the robot behavior in the video attached here.

[image]

Here is a video showing when we try to reach all the range of the robot right wrist prosup motion using motorgui while in reality the robot range of motion was much less:

https://user-images.githubusercontent.com/17707730/119855063-e444de80-bf11-11eb-9b2c-999f331c5d19.mp4

cc @DanielePucci  @violadelbono

kouroshD, 2021-05-28:
@violadelbono let me know if I should open an issue for the right thumb oppose and reopen the issue https://github.com/robotology/icub-tech-support/issues/1111

violadelbono, 2021-06-03:
Solved in https://github.com/robotology/icub-tech-support/issues/1111.
I saw you opened another 2 issues for the middle proximal and the thumb oppose, so I close this one.

closed 2021-06-03 by violadelbono