robotology/mesh-support #1138: Hole detected in in rx ropframe sequence number

S-Dafarra, 2021-05-26:
### Robot name 🤖

iCubGenova09

### Request/Failure description

When launching the ``yarprobotinterface``, several errors are detected regarding holes in the rx ropeframe sequence number.

@marcoaccame also noticed it in some other logs.

cc @DanielePucci @GiulioRomualdi @isorrentino 

### Detailed context

When starting the robot it seems that there are different errors regarding the ethernet connection.

### Additional context

Here some related log files.

[l_hip_pitch_encoder.txt](https://github.com/robotology/icub-tech-support/files/6548042/l_hip_pitch_encoder.txt)
[log_icub-head_yarprobotinterface_23384_hole.txt](https://github.com/robotology/icub-tech-support/files/6548043/log_icub-head_yarprobotinterface_23384_hole.txt)

marcoaccame, 2021-05-27:
The holes in sequence number are due to the presence of CRC errors. 

When the ETH switch detects a CRC error it drops the frame, the ems signals the CRC error and the software both on icub-main and the ems detects that the sequence number is not consecutive.

So, this issue is a twin of https://github.com/robotology/icub-tech-support/issues/1118.

pattacini, 2021-05-28:
Duplicate of #1118.

closed 2021-05-28 by pattacini