robotology/mesh-support #1139: Right thumb oppose does not calibrate as expected

kouroshD, 2021-05-28:
### Robot name 🤖

icubGenova04

### Request/Failure description

The right thumb oppose encoder feedback is high, so it is stays in HW fault. Moreover, the calibration is not done successfully.

### Detailed context

The right thumb oppose encoder feedback is high, so it is stays in HW fault

[image]

Moreover, in the calibration process the right thumb oppose reaches `0` degrees and therefore cannot smoothly calibrate the right thumb and touches the hand. Instead, the right thumb reaches the limit of 12 degs.

maggia80, 2021-05-28:
it could be due to a displacement of the hall effect sensor 2B3HO or a rotation of the magnetic ring. 

[image]

violadelbono, 2021-06-23:
Hi @kouroshD,

A deep maintenance on the right thumb has been performed:

- Tendon replaced
- Mc4PLUS board EB28 replaced
- Hall sensor of thumb oppose (joint 8) replaced + calibration

However, there is still a strange oscillation of the thumb oppose when moved with the motorgui, and we still need to address this problem (hardware/control ?). In order to avoid it, the only actual solution is to lower the PIDs of joint 8.

Uboldi80, 2021-08-05:
i tried to test the joint integrity but seems that the joint is ok.
unfortunately the problem is not caused by any slippage of the joint control chain (mechanical parts)

S-Dafarra, 2021-08-23:
This is still occurring. The right hand often fails to calibrate (plus we encounter some MAIS timeout from time to time). In particular, the thumb does not seem to move well when performing the initial calibration.

Uboldi80, 2021-09-06:
Friday i changed the MAIS board and with this replace i fix the Mais timeout.
furthermore, the emergency button was in short circuit.
The jack was replaced.
now i have one problem anymore on the EB9 in the right arm.

Uboldi80, 2021-09-10:
yesterday i replaced 1 MC4plus eb28 in the right arm, and i have fixed some problem with the can.
 now the situation has been improved but not solved.

Uboldi80, 2021-09-16:
After a deep maintenance in wich @emilianob80 has changed cover, skin and head supprts, I fix a lot of problems:
i Replaced  2Foc board under EMS EB9 in the right leg that doesent'work properly: pitch and roll of right leg doesnt'work but yarp motorgui started in a green mode.

I fixed with @emilianob80  fault line in the left arm that is broken in 3 different point.
I fixed the CAN on the right arm and I have largely solved the problem on the unstable thumb:

https://user-images.githubusercontent.com/80786976/133564346-1f33cd86-641b-4636-9f1a-352e4721f12f.mp4

S-Dafarra, 2021-09-16:
> After a deep maintenance in wich @emilianob80 has changed cover, skin and head supprts, I fix a lot of problems:
> i Replaced 2Foc board under EMS EB9 in the right leg that doesent'work properly: [...]

Great, thanks @Uboldi80 and @emilianob80!

> the palm skin power connector was unplugged and insulated with tape.
Is there a particular reason why it has been isolated? we left it like that

I have no idea :sweat_smile: 

Regarding the thumb, unfortunately, I think the modifications got lost. I guess you edited the files in the ``robotology-superbuild/build/install`` directory. These get completely overwritten when installing again ``robots-configuration``, and, unfortunately, this was the first thing we did today since we had to switch branch. Really sorry about this.

S-Dafarra, 2021-11-16:
I think this has been fixed. Right @Uboldi80 @GiacomoVassallo13?

Uboldi80, 2021-11-17:
I think could be ok.

S-Dafarra, 2021-11-17:
Closing then

closed 2021-11-17 by S-Dafarra