robotology/mesh-support #1140: Left middle proximal does not work

kouroshD, 2021-05-28:
### Robot name 🤖

icubGenova04

### Request/Failure description

The left middle proximal joint of robot icubGenova04 does not move.

### Detailed context

Here is a photo of the behavior from yarpmotorgui.

[image]

kouroshD, 2021-05-28:
> @kouroshD the joint doesn't move but in the motorgui it's green, right?

Yes, that's correct.

violadelbono, 2021-06-03:
> tests to be performed:
> 
> - check the connector
> - use the jogger for moving the motor

Checks performed and it moves smoothly. However, it requires about 4-5V to start moving. The other proximal joints move at around 1V so the motor is probably damaged.

violadelbono, 2021-06-22:
After inspection, the gearbox was damaged and often getting stuck.
It has been replaced and the joint has been cabled again. Now it works smoothly.

closed 2021-06-22 by violadelbono