robotology/mesh-support #1142: FT sensor stopped working, reports 0 N forces and 0 Nm torques on all axes

musghz, 2021-06-07:
### Robot name 🤖

SIPPC4

### Request/Failure description

Here is the problem we are facing. After we deployed one of the FT sensors with our custom robot, the sensor stopped sending force/torque values. Specifically, the sensor is giving 0 N forces and 0 Nm torques on all axes. It just started doing this about a week ago and has been the same since then, no matter how many times we power cycle it.

### Detailed context

**Our Application**
Here is the background for our usage of the FT sensors. We have been using the iCub FT sensors in our custom robot application for almost 9 years. The first batch was purchased in 2012. The new batch of sensors was purchased in 2020. It seems that these 2020 sensors have newer electronics and connector interface, but the same old communication interface...compared to the ones we got in 2012. This seemed to be case from the online documentation, and from brief testing that we did.

Now that we have deployed one of these sensors with a robot, the sensor has stopped working. It was working fine for a few weeks of testing.

**Observations**
Here are our observations. 0N and 0Nm is obtained after using the full scale conversion (obtained from the "get full scales" command). The values do not change no matter what force or torques are applied. We also tried use the "set transmission mode" command to 3 (transmit NOT calibrated data continuously). In this case, values at index 0, 1, 3, 4, 5 were changing based on applied forces and torques, but value at index 2 was not changing. It was constant at -32768. Perhaps this is indicative of a bad strain gauge that is not responding?

**Question**
Here is our question to you. Is this indicative of some software setting? Or is this indicative of a hardware issue with the sensor? We can do some more troubleshooting. We can also send the senor back to you for troubleshooting.

I am attaching a copy of the original FT sensor commands documentation that we received in 2013. This is what we have been using for the current sensors as well.
[FTSensor_user_manual.pdf](https://github.com/robotology/icub-tech-support/files/6610114/FTSensor_user_manual.pdf)

### Additional context

Here is how our code works in general. 

1. On power up, the transmission mode is set to 1 (do acquisition but do not transmit). 
2. Get the full scale data. 
3. The transmission mode is set to 0 (transmit calibrated data continuously). 
4. Then the incoming force/torque data are used by the robot. 

This is not a CAN bus issue as we have verified that the correct CAN packets with the correct CAN headers are coming in.

We are using an mbed GR-Peach microcontroller. It has been working fine for over a year with the older version of the FT sensor so we can say that there are no issues with that. Previously, since 2013, we have been using the mbed LPC1768 microcontroller with the 2012 batch of sensors.

maggia80, 2021-06-07:
hi @musghz, 

could you please tell us your ftsensor S/N?

musghz, 2021-06-07:
Hi @maggia80 

The serial number is 031, from the empty box. Would it be different on the physical sensor?

Here are other numbers from the box just in case:
12487.A3
FT45_M4_E2

maggia80, 2021-06-07:
yes, 031 is the S/N.

maggia80, 2021-06-08:
Hi @musghz, could you please test your setup with the other two sensors: S/N019 and S/N 048? It might be a failure in the S/N031, but we would like to be sure it's an issue related to this sensor.

musghz, 2021-06-08:
Hi @maggia80 
If you are wanting us to test the sensors, then we have done so on a copy of the robot that we have here. On this robot, we have been testing S/N 048 for a few days now. We also tested S/N 019 for a few minutes before shipping it on Friday, out to the deployed robot location, for replacement.

If you are particular about testing on that same robot which had S/N 031, we will find out within 12-24 hours. S/N 019 has reached the destination. Our team will remove S/N 031 and and install S/N 019 in its place very soon.

musghz, 2021-06-08:
Hi @maggia80 
Our team just removed the faulty sensor S/N 031 from the deployed robot, and installed S/N 019 in its place. S/N 019 is working fine, giving expected force and torque values. 

To sum up:
Robot no. 9, FT sensor S/N 031 stopped working, and S/N 019 is working (this is the deployed robot)
Robot no. 10, both FT sensor S/N 048 and S/N 019 are working (this robot has not yet been deployed)

Robot no. 9 and 10 are different copies of the same robot, the same in hardware and software.

One little thing from the quick test with S/N 019 is that the z-torque is flipped. We will do some more testing and know for sure.

musghz, 2021-06-08:
Hi @maggia80 

Just verified from further testing. On S/N 019 all three torques have the opposite sign. Is this a feature or a bug? It is not an issue for us as we have flipped the sign in software to compensate for this. But we are just curious to know.

musghz, 2021-06-08:
@pattacini Thanks for sharing this. This is interesting that your calibration of S/N 019 is fine but we are getting the flipped value after swapping in the sensor. We have checked pretty thoroughly but we can take another pass at the code. We had some debug code to troubleshoot but it is all out now.

pattacini, 2021-06-14:
Hi @musghz 

Do you have any news from your side?
The fact that you've been experiencing just the torques flipped in sign (and not the forces) for S/N 019 is puzzling.
(@simeonedussoni any idea on this?)

Regarding S/N 031, I believe you can send it back to us.

cc @maggia80 @andreaderito

ahfagg, 2021-07-05:
@pattacini As we were testing #019, we had a second robot with another functional sensor running at the same time.  We compared the forces and torques reported by the two in response to the same forces applied to the sensors.  Reported force directions matched, but torque directions were flipped.  We have since integrated a flip into the robot firmware for that one robot & the control system is behaving as expected.  From our perspective, this issue is solved.  But, it is worth finding an explanation.

ahfagg, 2021-07-05:
Coming back to sensor 031: how should we proceed with this one?  To recap: it worked fine at one point, but then started reporting force/torque of zero along all DOFs.  I looked at what the individual strain gauges were reporting under different loads.   Strain gauge #2 is reporting a constant -32768 (i.e., the raw bits that we receive for this strain gauge are all zeros).  I assume that under the transformation to calibrated forces and torques, this is resulting in a very large divisor, which gives us the zeros.  Is this a correct assessment?

pattacini, 2021-07-06:
Hi @ahfagg 

It might be that for some reason the values of the internal calibration matrix written in the EPROM got changed. It seems that we had spotted similar events: see https://github.com/robotology/icub-tech-support/issues/236.

It would help if we can have access to those values in order to compare them against the values we recorded within our internal report. To this end, we can provide you with the required instructions (cc @triccyx).

>Strain gauge no. 2 is reporting a constant -32768 (i.e., the raw bits that we receive for this strain gauge are all zeros). I assume that under the transformation to calibrated forces and torques, this is resulting in a very large divisor, which gives us the zeros. Is this a correct assessment?

It can be a good explanation yep; it very much depends on how the raw readouts interact with the calibration matrix.

>Coming back to sensor 031: how should we proceed with this one?

I would suggest that you get in touch with our manager @andreaderito <andrea.derito _ at_ iit.it>.

---

Just a minor note: the `#` symbol together with numbers creates [cross-links among issues](https://guides.github.com/features/issues/#:~:text=You%20can%20reference%20issues%20by%20typing%20in%20a%20hashtag%20plus%20the%20issue%20number.); hence, it's better to avoid using it for different purposes.

simeonedussoni, 2021-07-06:
> 
> 
> Coming back to sensor 031: how should we proceed with this one? To recap: it worked fine at one point, but then started reporting force/torque of zero along all DOFs. I looked at what the individual strain gauges were reporting under different loads. Strain gauge no. 2 is reporting a constant -32768 (i.e., the raw bits that we receive for this strain gauge are all zeros). I assume that under the transformation to calibrated forces and torques, this is resulting in a very large divisor, which gives us the zeros. Is this a correct assessment?

The only way to get all zero force/torques is to have a calibration matrix filled with zeros. 
Since the FT sensor can use two sets of calibration matrices, but we only flash one, it mwy be that at some time the device flipped is choice to the empty set of values. 
Regarding the ADC with values stuck to -32k, this indicates a hardware failure, but since ADC values are at numerator this should not result in ZERO output (and, at least, should give randomic output given any F/T value is a linear combination of all 6 ADCs)

simeonedussoni, 2021-07-06:
info gathered from @marcoaccame: when one or more of the ADCs get stuck "out of range" (which is defined between -31k and 31k) the FT sensor streams the last "useful" value, stored in registers. But upon initialization the registers are cleared and set to 0, thus the FT calculated willl be evenly zero.
Solution: send the FT back to IIT for maintenance

triccyx, 2021-07-06:
Speaking with @marcoaccame :

- About SN031 the sensor is probably damaged. If one ADC saturates the output will be zero (to be precise the last good value is returned in this case the init value that is zero) for **all force/torque**. How is the sensor damaged? It is possible that one of the resistors on ADC is burned. It should be repaired. @simeonedussoni if the matrix is all zero at least the added tare should be present. 
- About the "couple inverted" sensors. In this case, if possible we can analyze the conversion matrix in the sensor.
How we can do it? We need to download the matrix using CAN. @musghz, Have you a CAN-USB box and a Serial-CAN board/cable? In this case, I can send you the instructions. See the figure for example:
[image]

marcoaccame, 2021-07-06:
> About the "couple inverted" sensors. In this case, if possible we can analyze the conversion matrix in the sensor.
How we can do it? We need to download the matrix using CAN

As integration of the above, in here is a sequence diagram with the CAN message exchange used to retrieve the 36 entries of the calibration matrix from a ftsensor w/ address = 3. The host can be the CAN-USB box but also a generic MPU acting as a CAN node which is able to manage the messages in the diagram.

[image]

ahfagg, 2021-07-06:
Thanks, all!

I will get with Andrea about shipping 031 back.

I will also work on coding up a solution for our set up to extract the calibration matrix.  The robot is being used in Pennsylvania right now (and we are in Oklahoma), so it will take a bit of time to get an answer for you.

andreaderito, 2021-09-30:
Hi @ahfagg 
thank you for providing details of the parcel, we will soon organise the pick-up of the goods at the indicated origin. Iwill follow up shipping issue by email (while this issue will continue focusing on solving the problem of the sensor).
cc @pattacini @maggia80

pattacini, 2021-12-03:
We have recently received the sensor `IIT12487_SN031`. Thus, we did some preliminary tests reporting that we can connect to the sensor via `Firmwareupdater` and `STLink` although the sensor is not functional.

We suspect problems on the internal wires to the strain gauges. Further investigations are required.

pattacini, 2022-09-01:
Closing.
Feel free to open it up again if required.

closed 2022-09-01 by pattacini