robotology/mesh-support #1143: Cannot start R1: missing boards problem

randaz81, 2021-06-09:
### Device name 🤖

R1SN003

### Request/Failure description

On some occasions, yarprobtinterface fails to start due to missing boards.

### Detailed context

When this happens, we cannot ping boards that are after the board with the ip 10.0.1.12 in the chain (see picture below).
We thus think that a cable connection is somehow damaged. 
To investigate the problem, we removed the cover of the left arm, but visual inspection gave us no hints.

### Additional context

[image]

lnobile, 2021-06-14:
Robot seems to work now, the problem is in the 10.0.1.8 board that sometimes is not powered (we noticed this because of flickering led on the board). After moving the board with a finger it doesn't shut down anymore. It could be a flimsy cable issue but now seems to work, even after moving the board and connected cables.
I didn't closed the covers to let you investigate more deeply into the problem.

lnobile, 2021-06-14:
@Arya07

violadelbono, 2021-06-14:
Hi all,

There were 2 cables loose connected to board EMS12.
We fixed them and mounted the covers.

closed 2021-06-14 by violadelbono