robotology/mesh-support #1146: DNS not working on icub server?

matejhof, 2021-06-17:
### Device name 🤖

iCubPrague01

### Request/Failure description

We received the iCub setup this week and connected the server to the switch and the `icub-laptop`, `icub-head`, and `icub-cuda` to the switch. We run yarpserver on the `icub-laptop`.
The `icub-laptop` and the `icub-head` can ping each other. The `icub-cuda` cannot be pinged and cannot ping the others. Only when IP addresses are used, this is possible. 
It seems that the DNS server configured on icub-server is able to resolve just the icub-head and no other computers are being resolved. Tested with nslookup (server 10.0.0.1)
Also, when adding a new laptop to the switch, we cannot `yarp detect` without manually changing ip address in yarp config.
My colleagues @0xBEDA, @rozliv , @rustlluk may have more information.

### Detailed context

[image]

maggia80, 2021-06-17:
cc @mbrunettini

mbrunettini, 2021-06-17:
Can you confirm that 
1. the ethernet connection scheme (as described in the document shipped with the robot) has been followed in any step?
2. The setup was powered on following the startup procedure as described in the document shipped with the robot?

It seems that the DNS is not working and the most common root cause is when the` icub-server` is starting when the switch is not read and/or it's not connected properly

rustlluk, 2021-06-18:
I have double-checked all the cables and did the whole turning-on procedure again. Now `yarp detect` on connected laptops can properly find the `yarpserver` running on `icub-laptop`. However, the DNS server still knows only `icub-head` and `icub-srv`. But that's something we can probably fix by ourselves. I am just wondering what is the "best practice" used with _yarp_? Is it better to set names like `icub-laptop, icub-cuda` with DNS or to use IP addresses?

mbrunettini, 2021-06-18:
The DNS is configured to resolv both `icub-laptop` and `icub-cuda` in the same way, so I don't think the issue is on the DNS server saide.
Can you please do the following on `icub-head`
```
ping icub-laptop
ping icub-cuda
```

(of course both machines must be ON and connected with an **ethernet** cable to the `icub-switch`)

And post here the results? Thanks

The _best practice_ for you setup is to use the hostnames, BUT you can use IP as well because:
- `icub-head` has a fixed IP
- `icub-laptop` and `icub-cuda` has DHCP assigned IP, but with a reservation, so if yuo connect the with a cable they always get the same IP

The advantage in using _hostnames_ is that (if everything works correctly) you can always add another machine to the setup adn it will be reachable with its hostname, so you don't have to deal with IP. Plus, if you need to take `icub-latop` or `icub-cuda` OUT the setup (for any reason) it's already configured with the DHCP, which is the most common solution.

rustlluk, 2021-06-18:
Ok, great. Thanks for the info. Here the screens:
[image]
[image]

mbrunettini, 2021-06-18:
Thanks, but 

> Can you please do the following on icub-head

>

rustlluk, 2021-06-18:
I am sorry, my bad. I somehow read it incorrectly
[image]

mbrunettini, 2021-06-18:
Fine, now have to check the `icub-sever` : please ssh to `icub-server` (10.0.0.1) and issue the following commands
```
sudo systemctl status bind9
dhcp-lease-list
tar cvf /home/icub/named_log.tar /var/log/named/
```

and the please post here the results as well as the file `/home/icub/named_log.tar` created by the last command

Thanks!

rustlluk, 2021-06-18:
`Rustli-ubuntu` is my laptop (connect with Ethernet to switch), which is btw pingable from both `icub-laptop, icub-cuda`. 
[image]
[named_log.zip](https://github.com/robotology/icub-tech-support/files/6676220/named_log.zip)
*Github didn't let me upload tar, so you have it as zip

mbrunettini, 2021-06-18:
Apparently the DNS service is working correctly, though. This is also confirmed by the fact that you laptop is resolved by `icub-laptop `and  `icub-cuda`
I think this deserves a live debugging session, maybe we should arrange a call with the help of @violadelbono or @Uboldi80 .

Meanwhile, it can be useful to know:

1. (while connected to the `icub-switch`) from `Rustli-ubuntu` the output of `ping google.com`
2. (same as above) from `icub-laptop` the output of `ping google.com`
3. (same as above) the ip address of  `icub-laptop` (eg. with the output of `ip addr`)

Thanks again.

rustlluk, 2021-06-18:
Pinging Google gives the same output on both machines
[image]
IP address is 10.0.0.111 (corresponds to the label on USB-Eth adapter)
[image]

I tried to plug ethernet on `icub-laptop` directly to the laptop and it successfully obtains IP from DHCP -- a different MAC address than the USB adapter, so the static reservation does not get activated. It made me try to disable the reservation of both laptops in `/etc/dhcp/dhcpd.conf` on `icub-srv`, which made both laptops resolved by DNS. But after enabling the reservation again the old problem arises.  

Is it important to have the static reservation for the laptops (they even get the right addresses without reservation, but I am not sure if it's a permanent thing or just some temporary result from the previous reservation)? If the reservation is needed and you would like to do the live debugging, I am up for it.

Thanks!

mbrunettini, 2021-06-18:
Please **always** use the USB adapter on the icub-laptop, as written in both documentation and on the laptop itself.

the reservation is important because _at the moment_ robot-bazaar does not support hostnames (but we are working to fix it), so we reserved  IPs, so they can be used with robot-bazaar.

mbrunettini, 2021-06-18:
On `icub-server`, can you please 
1. backup the file `/etc/dhcp/dhcpd.conf` with
```
sudo cp /etc/dhcp/dhcpd.conf /etc/dhcp/dhcpd.conf.orig
```
2. edit the file
 `/etc/dhcp/dhcpd.conf`
by replacing the following line
```
#allow client-updates;
```
with
```
allow client-updates;
update-static-leases true;
```

(i.e. you should uncomment the line `update-static-leases true;` and add the line `update-static-leases true;` )

and the restart the dhcpd service with
```
sudo systemctl restart isc-dhcp-server
```

And let us know, thanks!

rustlluk, 2021-06-18:
It seems to fix the issue, both laptops are now pingable and `nslookup` knows the names even when the devices are offline. 

I think we can close the issue.  @matejhof 

Thanks a lot!

closed 2021-06-18 by rustlluk