robotology/mesh-support #1149: iCubGenova01 - skin not responding as expected

xEnVrE, 2021-06-23:
### Device name 🤖

iCubGenova01

### Request/Failure description

The skin is not responding as expected (not activating on all the surface or not responding at all).

### Detailed context

The following parts have some patches that are not activating as expected:
- torso
- right forearm

The following parts are not activating at all:
- right hand palm
- right hand fingertips

violadelbono, 2021-07-05:
> - right forearm
> - right hand palm
> - right hand fingertips

Some skin connectors were faulty and have been restored. Moreover, the wiring of the thumb fingertip R0 was soldered incorrectly on the board. It has been also replaced.

Uboldi80, 2021-08-31:
the chest skin was damaged in several places
some cables were torn and a portion of skin w0mash0.
thanks to @emilianob80 to replace the mash.

closed 2021-08-31 by Uboldi80