robotology/mesh-support #1150: iCubGenova01 - right eye camera not working in some configurations

xEnVrE, 2021-06-23:
### Device name 🤖

iCubGenova01

### Request/Failure description

The right eye camera does not work in some configurations of the eye bulb.

### Detailed context

The right eye camera does not work in some configurations of the eye bulb. In such cases, the image goes almost black with weird blue stripes. See the image attached in the `Additional Context` section.

### Additional context

[image]

violadelbono, 2021-06-25:
The flat cable of right camera was damaged and has been replaced.

closed 2021-06-25 by violadelbono

xEnVrE, 2021-06-25:
Thank you @violadelbono @Uboldi80