robotology/mesh-support #1151: Neck motor hardware fault

giulia-belgiovine, 2021-06-28:
### Device name 🤖

iCubGenova40

### Request/Failure description

When using the robot head joints 0 and 1 goes in hardware fault. When manipulating the motor by hands it get stuck. Probably problem in the harmonic motor of the head.

### Detailed context

When starting the robot and moving the head by motorgui Joint 0 and 1 goes to Hardware fault.

pattacini, 2021-06-28:
Hi @giulia-belgiovine 

Could you better specify the name of the robot?
`iCubGenova40` does not exist (yet) 😄

giulia-belgiovine, 2021-06-28:
Sorry for the mistake! I updated the robot name :) Thank you!

pattacini, 2021-06-28:
Never mind!

violadelbono, 2021-07-06:
Ciao @giulia-belgiovine,

As discussed, after Julien intervention yesterday, we glued the washer to the little gear of the left side of the neck. Moreover, since we didn't have a M2x6 pin, we filed the M2x10 which was mounted. Now it does not interfere with the belt movement.

closed 2021-07-06 by violadelbono