robotology/mesh-support #1152: iCub sholder yaw joint fault !

ivan9a, 2021-06-30:
### Device name 🤖

iCub, Zagreb, Croatia, SN: 045

### Request/Failure description

Hi,

We have major malfunctions with yaw joint on the left arm, please look at the pictures down below:

[image]
[image]
[image]
[image]
[image]
[image]

We have only run your redball demo, during initialization one steel cable just broke and everything suddenly stopped. Do you have any instructions on how to replace steel cables ? Can you provide new or replacing parts for that fault joint ?

Thank you.

Best regards,
Ivan Zupancic

### Detailed context

One of steel cable broke during redball demo demonstration.

violadelbono, 2021-06-30:
Hi Ivan,

We will send you the cables and you can follow our online documentation to repair them.

First, of all, looking at the following picture, can you identify exactly the broken one(s)?

Thnks,
Viola

 
[image]

ivan9a, 2021-07-01:
Hi Viola,

Cables one and eight are broken, cables two and seven are damaged. It would be perfect if you can send us all four cables. We also need plastic inserts which are mounted on motor aluminum wheel, these plastic inserts prevent steel cables from stucking inside channels which are made for tensioning the cables. We can continue communication via e-mail if you need my post data.

Thank you,
Ivan Zupancic

violadelbono, 2021-07-02:
Hello Ivan,

Thank you. Let's continue this by email so we can arrange the shipping.
We will let you know soon. When the cables will be restored I will close this issue.

ivan9a, 2021-07-02:
Hello Viola,

Ok our shipping address is: Ivana Lučića 5, 10000 Zagreb, Croatia, Faculty of mechanical engineering university of Zagreb, Croatia.
Contact person: Ivan Župančić, phone: +385996792639

It would be perfect if you can send us:

1) steel cables one, two, seven and eight.
2) Plastic inserts which are mounted on motor aluminum wheel.

if we have to pay something feel free to send us an proforma invoice,
VAT ID HR22910368449

Thank you,
Best regards,
Ivan Župančić mag.ing.mech
Researcher,

ivan9a, 2021-07-02:
Hello Viola,

Thank you for your fast reply, ok please send me mail so we can exchange post data, my official mail is: ivan.zupancic@fsb.hr 

Thank you,
Best regards 
Ivan Zupancic

ivan9a, 2021-07-09:
Hello Viola,

I didn't receive any mail from you, can you give me any contact it is urgent for us. 

Thank you,
Best regards,
Ivan Župančić

Uboldi80, 2021-07-09:
Hi Ivan
Yesterday was Viola's last day of work.
As for the shipment, I have just urged the people who are in charge of shipping the spare parts, which at the moment are ready to go.

ivan9a, 2021-07-09:
Hi Uboldi,

Thank you for information, if you have some questions about shipping adress feel free to conntact me.

Best regards,
Ivan Župančić

andreaderito, 2021-08-06:
Hi Ivan,
apologise for my late feedback, we're ready to ship the parcel, I will use the following address:
University of Zagreb
Faculty of Mechanical Engineering and Naval Architecture
Ivana Lučića 5, eastern building,
10000 Zagreb, Croatia
Ref. person: Ivan Župančić
Tel: 00385 01 6168575
Pls let me know if any change is needed, thx in advance and best

andreaderito, 2021-08-12:
Hi @ivan9a and all, as you know ((ref. to email exchange) the parcel with the spares have been received yesterday 11.08.2021, they are in the hands of your colleague Boris. Best reagrds.
cc @pattacini @maggia80 @Uboldi80 @Fabrizio69

ivan9a, 2021-08-16:
Hi Andrea,
Thank you once again for sending me spare parts, I receive parts from my colleague Boris. Today I will mount new parts on the robot. I hope that everything will be fine. Best regards

ivan9a, 2021-08-16:
Hi Andrea,

Today I tried to fix the cable problem on the robot but unfortunately, I could not remove cable number 2. 
Should I remove cables 3 and 4 first and then tried to remove cable 2? Do you have some instructions on how to properly remove cables 2 and 1? And do you have some instructions on how to install them?

[
[image]
](url)

Thank you,
Best regards,
Ivan Župančić

maggia80, 2021-08-16:
Hi @ivan9a ,

You should follow the documentation here:

https://icub-tech-iit.github.io/documentation/icub_tendons/shoulder_elbow/

ivan9a, 2021-08-17:
Hi Maggia80

Thank you for the documentations. I have successfully managed to replace all broken cables, I also tested the shoulder joint, and now everything works fine.
[image]
[image]

Thank you,
Best regards,

closed 2021-08-23 by Uboldi80