robotology/mesh-support #1154: iCubGenova01 - Joints from l_wrist_prosup onwards goes in idle during arm motion

xEnVrE, 2021-07-01:
### Device name 🤖

iCubGenova01

### Request/Failure description

While executing the red ball demo, we experienced an issue with the left arm going partially in idle.

### Detailed context

While executing the red ball demo, we experienced an issue with the left arm going partially in idle. Specifically, all the joints from the `l_wrist_prosup` onwards (up to the left pinky finger) goes in idle unexpectedly.

I am attaching a video where the issue is clearly evident because the wrist goes down, being uncontrolled, due to gravity.

https://user-images.githubusercontent.com/6014499/124162037-5cb53700-da9e-11eb-9a66-2e7a4e1bae37.mp4

After the issue occured, we opened the `yarpmotorgui` and we verified that all the joints from the wrist prosupination onwards were in in idle.

We repeated the experiment three times, with the same outcome. We remark that the red ball demo is not directly linked with the issue (probably), while the fact of moving the arm in general is.

maggia80, 2021-07-02:
@xEnVrE did you save the log of yarprobotinterface? It would be important to see if there is any External Fault error there. @violadelbono @Uboldi80 it could be due to a cable connection issue on the fault wire at the upperarm level. 

[image]

page 5 of https://github.com/icub-tech-iit/electronics-wiring-public/blob/master/icub2/icub-2.0.0/pdf/iCub%202_2.0.0_XXXX_Logic.pdf

xEnVrE, 2021-07-02:
@maggia80 no log unfortunately due to https://github.com/robotology/yarp/issues/2630#issuecomment-872379579.

However, the issue is repeatable, hence I can do it again while temporarily disabling the feature that was causing so many messages in the log (as per the aforementioned issue).

Uboldi80, 2021-07-02:
the problem with the left arm calibration is fixed.
I had to redo all the fault connectors on the left arm (included P28 and P28J) and replace the 1B2 board

closed 2021-07-05 by violadelbono