robotology/mesh-support #1156: iCubGenova01 - Left thumb opposition encoder misbehaving

xEnVrE, 2021-07-05:
### Device name 🤖

iCubGenova01

### Request/Failure description

The encoder of the joint left thumb opposition does not work.

### Detailed context

The encoder of the joint left thumb opposition does not work. Specifically, it provides a sort of fixed value (~ -120.0 degrees) independently from the actual position of the joint.

Uboldi80, 2021-07-05:
Hal'senor of thumb oppose appears to have broken.
probably need to change.
We tried to test the sensor with the jogger but doesn't work

violadelbono, 2021-07-07:
Hi @xEnVrE,

We found ot that the problem was actually the connector P8 on the board 1B3 (Hall sensor of thumb) which was not working anymore:

[image]

Changing that board solved the problem.

closed 2021-07-07 by violadelbono

xEnVrE, 2021-07-07:
Thank you @violadelbono @Uboldi80