robotology/mesh-support #1158: Icub robot does not want to turn on

ivan9a, 2021-07-09:
### Device name 🤖

iCubZagreb01

### Request/Failure description

Icub robot does not want to turn on 

### Detailed context

Icub robot does not want to turn on, when I press the power button and motor driver button nothing happens. In normal conditions, these buttons should be in green light.
Please see pictures down below:

[image]

[image]

Thank you,
Best regards,
Ivan Župančić

Uboldi80, 2021-07-09:
Hi Ivan
from what i see, the leds do not even turn red, therefore, no power arrives to the robot.
please check the following:
- verify the cables of power supply, from the power supply to the robot (maybe are loose something to the connection?) and take a photo please.
- Verify the connector cable under the backback and check if is thightened 
- if the previous checks failed, check the fake battery in the back pack and try to press lightly on it

let me kno

ivan9a, 2021-07-09:
Hi Uboldi,

The LEDs do not turn red but LEDs inside the robot turn on, I have checked every connection as you described. Please see the pictures down below:

[image]
[image]
[image]
[image]
[image]

The connector under the backpack is tightened, everything looks ok with cables.

Thank you,
Best regards,
Ivan Župančić

Uboldi80, 2021-07-09:
OK, as far as I can see the voltage reaches the board and in fact the LED are on.
you should check on the BCB (the board that houses the power keys that do not light up) below the image that indicates the point to check.

[image]

Uboldi80, 2021-07-09:
you should check that the pins are not bent and consequently NOT TOUCH each other, otherwise this could reset the board and give this problem.

Uboldi80, 2021-07-09:
hello Ivan

to perform this check I suggest you:

turn off the setup, and disconnect the power cable and the ethernet cable

Uboldi80, 2021-07-09:
if due to the backpack you cannot see that point on the card then I will show you the steps to remove the battery and the backpack.
Unfortunately, those pins are very close to the plastics, and may have bent due to a bump or pressure at that point.

ivan9a, 2021-07-09:
Hello Uboldi,

The problem is fixed !, I have removed the plastic and checked all pins on that PCB. The first two pins were connected, now everything is as it should be.

Thank you,
Best regards,
Ivan Župančić

Uboldi80, 2021-07-09:
Great!!!! 

thankyou @MrAndrea, master stroke!

Uboldi80, 2021-07-09:
ok Ivan 

i'm very happy for you, and now i will close this issue.

closed 2021-07-09 by Uboldi80