robotology/mesh-support #1159: Broken connector for torso sensor

cedricgoubard, 2021-07-13:
### Device name 🤖

iCubLondon01

### Request/Failure description

Hello,
Our iCub has a broken connector near the torso sensor:

[image]

Do you have any advice on how to fix it? Should I order a new connector, or try to solder it? Could you tell me where to find the connector reference/the wire size?

Thank you,
Cedric

### Detailed context

From [the schematics](https://media.githubusercontent.com/media/icub-tech-iit/electronics-wiring-public/master/icub2/icub-2.0.0/pdf/iCub%202_2.0.0_XXXX_Logic.pdf), I think it is the incoming connector to CANFilter_B (CFB18, in the middle on the picture):

[image]

Uboldi80, 2021-07-13:
Hy Cedric

Can you say me the label of that cable? by the photo seems N9?

Instead...to solve this problem we can move in 2 steps:

1 if the connector broken stop your activity, you can try to fix with solder
2 the better way you do, is order the replace parts:

- the molex crimp inside the connector have the code 8 in our system, and the supplier is RS components, RS code is: 670-6414
and the Molex code is: 50058-8000

- The Molex 4 positions connector is 9 in our system, and the supplier is RS components, RS code is: 279-9162 and the Molex code is: 51021-0400

- To crimping the wire, you need to have the Molex pliers: The supplier is Farnell and the Farnell code is: 2837322 and the Molex code is 	63827-1500

Uboldi80, 2021-07-13:
To remove the crimp inside the connector, you should "open"the pin locker.
the sequel image is rappresentative:
with a cutter blade you can "open"the locker and extract the crimp to sold it with the wire.
[image]
after this action you can extract the crimp:
[image]
after having soldered the crimp to the wire and put them back in the connector, press the pin locker so that it can lock the cable in the connector itself.

Uboldi80, 2021-07-13:
so, if you give me the cable label we can check if is possible make a new cable and send it to you, 

best regards

cedricgoubard, 2021-07-13:
Hi @Uboldi80,
Thank you for the explanation; we managed to fix the connector based on your instructions!
All the best

closed 2021-07-13 by cedricgoubard