robotology/mesh-support #1160: iCubErzelli02: Left hand little fingers not working

ddetommaso, 2021-07-15:
### Device name 🤖

iCubErzelli02

### Request/Failure description

The little fingers of the left hand are not working properly.

From yarpmotorgui the home position of the joint 15 is 2.7 instead of 0. Sliding the cursor to higher values of position the target position is not reached, so the fingers do not close.

### Detailed context

26.790468	ERROR		 from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 874s 111m 191u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .

Uboldi80, 2021-07-15:
Hy Davide
i think could be a tendon broken, and then, i will replace in the next week.
so if is good for you, i will be wednsday in erzelli.
let me know

ddetommaso, 2021-07-15:
Hi @Uboldi80 

Wed it's fine with me!
@vtikha does it sound good for you and your team as well?

vtikha, 2021-07-16:
Ok for us 👍🏼

Uboldi80, 2021-07-21:
The tendons of joint 15 were replaced and the operation was successful.
The anterior chest support has also been repaired.
A condenser was attached to the lower front of the chest.
The robot has been tensioned and is working.

closed 2021-07-21 by Uboldi80