robotology/mesh-support #1161: R1SN003 - left arm roll joint not working properly

lnobile, 2021-07-15:
### Device name 🤖

R1SN003

### Request/Failure description

R1SN003 - left arm roll joint not working properly

The arm movement are not smooth and the robot enters in over current protection as if something in the mechanics is not rotating smoothly.

### Detailed context

A video demonstrating the problem:

https://user-images.githubusercontent.com/56638910/125784370-46b81ff3-d510-4d43-8e46-8a6f00aacf86.mp4

Uboldi80, 2021-07-16:
After a deep intervention of maintenance on the shoulder we make a solution.
@fbiggi @elio75 with my support make lot of mdify:

- increase the space for a shoulder gear with the dremel
- make 8 shims to decrease the axial movement of the shoulder gear
- milling the plate that support the shaft (1mm)
 I think that the robot is ready to start in Monday morning

Elio75, 2021-07-16:
The fact that the gear wheel turns eccentrically helped even more to nail the worm, so we put the gear wheel on the side where the eccentricity was lowest in order to help the system not to nail

fbiggi, 2021-07-19:
@lnobile I deliver r1 assembled
we await your ok to close the issues

@Uboldi80

Uboldi80, 2021-07-20:
https://user-images.githubusercontent.com/80786976/126337470-9a405fc4-aa36-4aca-953a-f5d376375529.mp4

the shoulder is ok!

closed 2021-07-20 by Uboldi80