robotology/mesh-support #1163: iCubGenova01 - Cameras @ 640x480 stop working after a while

xEnVrE, 2021-07-19:
### Device name 🤖

iCubGenova01

### Request/Failure description

The cameras of the iCub eyes stop working after a while (15 to 20 minutes) when run at 640x480.

### Detailed context

The cameras of the iCub eyes stop working after a while (15 to 20 minutes) when run at 640x480. When this happens, restarting the associated `yarpdev` does not help as either it runs successfully (but the image streaming does not start) or it returns an hardware error at startup (I will copy it here asap).

The only solution is to restart the `PC104`.

pattacini, 2021-07-20:
It could be due to a nasty memory leak.
During normal operations, it'd be helpful to monitor the memory allocated by the corresponding `yarpdev` services by means of for example the command `top`.

cc @xEnVrE

xEnVrE, 2021-07-20:
Thanks @pattacini 

I will try it and let you know.

triccyx, 2021-07-20:
> It could be due to a nasty memory leak.
> During normal operations, it'd be helpful to monitor the memory allocated by the corresponding `yarpdev` services by means of for example the command `top`.
> 
> cc @xEnVrE

As a restart of yarpdev doesn't help I fear the problem can not be there.
But after yarpdev restart, I think you should also reconnect the yarp port from yarpdev to yarpview for example

pattacini, 2021-07-20:
>As a restart of yarpdev doesn't help I fear the problem can not be there.

Correct 👍🏻 
I missed this point.

sgiraz, 2021-07-20:
I suggest also `htop` which is installable through `apt` and it provides the same information but graphically more intuitive.
If you have the possibility to build the code with debug symbols, you may run it using [Valgrind](https://valgrind.org/). If a memory leak exists, it will be found.

triccyx, 2021-07-20:
The output from the shell where you start yarpdev after the `malfunction` can help. Copy and past the last lines.

xEnVrE, 2021-07-20:
Thanks everyone for all the suggestions!

>  it'd be helpful to monitor the memory allocated by the corresponding yarpdev services by means of for example the command top

the `yarpdev` stabilizes on `1.5%` of the available RAM. I didn't see any uncontrolled increase of RAM utilization.

> The output from the shell where you start yarpdev after the malfunction can help. Copy and past the last lines

Unfortunately, when the issue happens the `yarpdev` continues printing the usual message like `the device is running in background`.

After the last tests I realized that:
1. the issue seems to present only if the `yarprobotinterface` is running
2. I was not using the right camera before, now I realized that the **only camera that stops streaming is the left one**. I will update the title of the issue then. This seems to confirm a problem associated to the device.

pattacini, 2021-07-20:
Hi @xEnVrE 

@vtikha and @AlexAntn have reported that it might be the temperature also in this case.
In particular, @AlexAntn observed this with iCubPlymouth01. Removing the covers did help in that case.

pattacini, 2021-07-20:
Given that @xEnVrE is working with iCubGenova01 in view of the important demo of G20 next week, I think it'd be worth getting ready to replace that particular dragonfly.

Uboldi80, 2021-07-27:
I changed the flat cable of the left camera (code 356)
the old flat was badly damaged.
@xEnVrE  please let me know if it solves the problem.

xEnVrE, 2021-07-27:
We re-tested both cameras @ 640x480 and they were running properly for more then one hour with the `yarprobotinterface` running.

I think we can close the issue.

xEnVrE, 2021-07-27:
Thank you @Uboldi80

closed 2021-07-27 by xEnVrE