robotology/mesh-support #1164: Left shoulder AEA encorder streaming invalid data

gabrielenava, 2021-07-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

Sometimes while the robot is position controlled, the left arm shoulder joints go in hardware fault. From the logger we sse there is a problem with the AEA encoder:

### Detailed context

```
[ERROR]  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 900s 675m 89u: (code 0x02000008, par16 0x0002 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

### Additional context

The error does not occur always, therefore it may be cause by the encoder cable and not by the encoder itself.

cc @DanielePucci @maggia80 @Fabrizio69

DanielePucci, 2021-07-21:
Infinite thanks to @maggia80 that helped us fix the issue, thanks a lot Maggia

@robotology/iit-dynamic-interaction-control

Uboldi80, 2021-07-21:
the problem was solved by replacing a crimp of a connector.

thankyou @maggia80 !!

closed 2021-07-21 by Uboldi80