robotology/mesh-support #1165: iCubGenova04 - AEA encoder issues on right ankle, both joints

gabrielenava, 2021-07-21:
### Device name 🤖

iCubGenova04

### Request/Failure description

During the iRonCub experiments we experienced hardware fault on the right ankle joints, both pitch and roll. The error message seems related to the AEA encoder.

### Detailed context

Here the error message (there was a similar one for the other ankle joint):

```
[ERROR]  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 2970s 405m 204u: (code 0x02000008, par16 0x0001 par64 0x000000000000000a) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

### Additional context

Despite the error message, we believe the problem may be also related to high temperature as the foot is close to the exhaust cone. Also, the problem seems not very repeatable at the moment. Nevertheless if someone is available tomorrow it would be nice to check the status of the ankle.

cc @DanielePucci

Uboldi80, 2021-07-26:
This morning I checked the board part related to the ankle.
I did not find anything abnormal, so I could assume that the fault was given by the high working temperatures (60 degrees)

gabrielenava, 2021-08-05:
Thank you @Uboldi80, in the remaining days of experimental campaign this problem did not happened again. I think we can close the issue for now, if the problem will occur again I will open a new one.

Uboldi80, 2021-08-24:
Hy @gabrielenava 
do you think is possible close this issue?
THKS!

gabrielenava, 2021-09-06:
>Hy @gabrielenava
do you think is possible close this issue?
THKS!

sure @Uboldi80, I close it!

closed 2021-09-06 by gabrielenava