robotology/mesh-support #1166: iCubGenova01 - Left index proximal joint oscillates midways

xEnVrE, 2021-07-24:
### Device name 🤖

iCubGenova01

### Request/Failure description

When going beyond 30/40 degrees, the left index proximal joint starts to oscillate. The readings from the yarpmotorgui suggest a problem in the feedback from the encoder.

Please find attached a video showing the behavior.

### Detailed context

When going beyond 30/40 degrees, the left proximal index joint starts to oscillate. The readings from the yarpmotorgui suggest a problem in the feedback from the encoder.

Please find attached a video showing the behavior.

### Additional context

https://user-images.githubusercontent.com/6014499/126873694-dd207ff0-49e8-4b0c-b9c3-86042f66dc6c.mp4

Uboldi80, 2021-08-05:
I moved it and unlocking it, the joint is back in operation.
I tested it with the motorgui and it no longer has problems of instability
if the problem arises again, the joint reducer 11 will probably have to be replaced

xEnVrE, 2021-08-25:
Ciao @Uboldi80,

I tested it again this morning and the problems persists.