robotology/mesh-support #1167: iCubErzelli01: issue with joint 13 (left-arm)

ddetommaso, 2021-07-27:
### Device name 🤖

iCubErzelli01

### Request/Failure description

Altough the calibration process is correctly executed the joint 13 (left-arm) remains in NOT CONFIGURED mode with encoder value around 7.0.

Attached the log session 

[yarprunlog_27_07_2021_09_35_13.log](https://github.com/robotology/icub-tech-support/files/6883570/yarprunlog_27_07_2021_09_35_13.log)

### Detailed context

-

pattacini, 2021-07-31:
I think that the relevant line in the log is:

```console
embObjMotionControl::checkRemoteControlModeStatus() has done 2 attempts and will retry again after a 0.010000 sec delay. (BOARD left_arm-eb26-j12_15 IP 10.0.1.26, joint 1) -> current mode = cal, requested = idl
```

The board tries to change the mode of joint 13 (with several attempts in a row) but it doesn't come off.
Probably, the calibration of that joint doesn't complete successfully and the process hangs up unexpectedly.

I didn't find the reason for that in the log yet.

@marco-monforte @ale-git @davidetome any clue as to why this may happen?

---

Hi @ddetommaso 

The log is weirdly formatted, maybe because it comes from the `yarprun` and not from the `yarplogger`; the latter is definitely a better choice.

pattacini, 2021-07-31:
Line 15087:

```console
from BOARD 10.0.1.26 (left_arm-eb26-j12_15), src LOCAL, adr 0, time 304s 864m 902u: (code 0x02000010, par16 0x0003 par64 0x0000000000000000) -> MC: hard limit reached. The joint position is outside its hardware boundaries. par16 = ID of joint. + .
```

This line comes right ahead of the warnings above and might be related to the problem although it's not clear to me if it refers to the joint 1 of the board (i.e. joint 13).

There are other similar lines in the log.

marco-monforte, 2021-07-31:
@pattacini maybe you wanted to tag @marcoaccame, not me :smile:

pattacini, 2021-07-31:
That's correct!

Uboldi80, 2021-08-02:
I think probably need to recalibrate the joint

Uboldi80, 2021-08-04:
Today i noticed 3 different problems on this setup:

- the right side of the robot, from the torso, to the right arm was off.
- the encoder R2S2 doesn't work
- the joint 13 left hand was in UNCONFIGURED STATUS
The first problem is solved by fixed the wire broken in EB5
The second problem was fixed by replace of R2S2 connector (to MC4 BOARD)
And the third problem (joint 13 non configured status) it was created by an overcurrent status.
the joint 13 was blocked due to dirt on the pulley, and them the board put the joint in a safety status mode trough the "non configured joint"

All the setup it was tested with @ddetommaso  and was fine!

closed 2021-08-04 by Uboldi80