robotology/mesh-support #1168: Right hip pitch cable broken

gabrielenava, 2021-07-27:
### Device name 🤖

iCubGenova04

### Request/Failure description

the right leg hip pitch joint metal cable got broken during an experiment.

### Detailed context

while performing experiments, the robot landed violently on the ground and the hip pitch cable broke.

maggia80, 2021-07-27:
@Uboldi80 the documentation related to that tendon is here: https://icub-tech-iit.github.io/documentation/icub_tendons/waist_legs/#31-hip-flexion-cable-14

@fbiggi knows how to fix it, once the electronics cards have been removed.

Uboldi80, 2021-08-03:
Today @fbiggi replace the tendon broken.

fbiggi, 2021-08-03:
riconsegno il robot in dara 03/08/2021 tempo di intervento 3 h 1/2

[image]
[image]

closed 2021-08-03 by Uboldi80