robotology/mesh-support #1169: iCubGenova01 - Left camera images contains artifacts

xEnVrE, 2021-07-30:
### Device name 🤖

iCubGenova01

### Request/Failure description

When running the **left** camera for a while, strange artifacts start appearing in the top part of the image.

https://user-images.githubusercontent.com/6014499/127656318-a449c419-2df7-4a45-9197-bd333e2276fa.mp4

### Detailed context

When running the **left** camera for a while, strange artifacts start appearing in the top part of the image.

traversaro, 2021-08-05:
We had some similar problems related to the `libdc1394` library on other setups,  evein if in that case the problem was highly non-deterministic, and was disappearing if the `yarprobotinterface` was launched before hand. Just to understand, in this case the `yarprobotinterface` was running or not?

xEnVrE, 2021-08-05:
In this case the `yarprobotinterface` was running.

Uboldi80, 2021-08-06:
@traversaro, do you think that problem could be related an hardware problem? ccd of left camera?

traversaro, 2021-08-23:
> 
> 
> @traversaro, do you think that problem could be related an hardware problem? ccd of left camera?

I don't have a clear idea, but @maggia80 indeed suggested that that could be the case.

maggia80, 2021-08-25:
yes! I think so since the corrupted image does not have rows with issues but it's a more weird behavior. We could try to replace just the CCD.

Uboldi80, 2021-08-25:
ok @maggia80  i can try to replace that.

Thks!

Uboldi80, 2021-08-31:
Today I replaced the CCD of the left camera.
for now seem to work well.

xEnVrE, 2021-08-31:
The CCD has been replaced but the problem still persists at 640 x 480.

Nevertheless, since it is not happening at 320 x 240, that is the usual target when using this robot, I think we can close this for now. 

This also means that the problem might be:
- hardware related, as it presents only when running at higher resolution, maybe related to the firewire bandwidth, etc?
- software related, maybe one of the modules inside the chain from the camera to the output is broken (maybe a memory leak?)

closed 2021-08-31 by xEnVrE