robotology/mesh-support #1170: Fingertip taxel coordinates (3D relative to *_tip frame)

maxymczech, 2021-07-30:
### Device name 🤖

iCub

### Request/Failure description

I would like to know where I can get 3D coordinates for the skin taxels on the tips of iCub robot's fingers.

### Detailed context

Hello. I apologize for creating this issue in `HW Support Request`. I have been working with iCub robot model in Gazebo 11 simulator, trying to add artificial skin sensing capabilities. I have installed `icub-main` and `icub-models`, and I have found taxel coordinates for most body parts in folder `/usr/share/iCub/contexts/skinGui/positions/`:

```
left_arm_mesh.txt
left_forearm_mesh.txt
left_forearm_nomesh.txt
left_forearm_V2.txt
left_hand_V2_1.txt
right_arm_mesh.txt
right_forearm_mesh.txt
right_forearm_nomesh.txt
right_forearm_V2.txt
right_hand_V2_1.txt
torso.txt
```

However, the `hand` files contain only coordinates of the taxels on the palm, no fingertips included. I would be really thankful if you pointed me in the right direction here.

pattacini, 2021-07-30:
Hi @maxymczech 

This is actually not the correct place for such a request as per the [guidelines](https://github.com/robotology/icub-tech-support/blob/master/.github/SUPPORT.md).
Could you please file a discussion in our https://github.com/robotology/community instead?

traversaro, 2021-07-30:
> Hi @maxymczech
> 
> This is actually not the correct place for such a request as per the [guidelines](https://github.com/robotology/icub-tech-support/blob/master/.github/SUPPORT.md).

Sorry, I probably suggested was the one that suggested the wrong place to @maxymczech . In any case, thanks for opening the issue @maxymczech , and please follow @pattacini advice, thanks!

pattacini, 2021-07-30:
Hi @maxymczech 

I've created the discussion in https://github.com/robotology/community/discussions/538.

Closing this then.

closed 2021-07-30 by pattacini