robotology/mesh-support #1171: Replacement options for DragonFly cameras

cedricgoubard, 2021-08-03:
### Device name 🤖

iCubLondon01

### Request/Failure description

Hello,
The right Dragonfly camera on our iCub broke, and we are considering several options to replace it.

### Detailed context

Here are our ideas:
- **Get the exact same one** (DR2-03S2C-EX-CS) - Could we order a new one from the iCub team, or should we contact the manufacturer?
- **Get a newer version** - do the new iCubs come with a better camera that we could mount on ours?
- **Replace it using Raspberry Pi** - the Raspberry Pi cameras would fit in the eyes, and the Raspberry Pi themselves could probably fit in the place of the Dragonfly cards.

What solution would you recommend?

Uboldi80, 2021-08-06:
Hello Cedric, we have analyzed your problem, and we have decided on the following option.
You will be contacted by email by my collegue mr. Derito to receive an offer regarding the replacement of the Dragonfly and the flat cable attached to it.
The replacement of the Dragonfly, involves the recovery of the lens and the lens holder from your current setup.

cedricgoubard, 2021-08-06:
Hello Paolo,
That sounds good! Here is my email address: c.goubard20@imperial.ac.uk
Regarding the recovery of the lens & lens holder, is there any documentation I could use?

Uboldi80, 2021-09-14:
Hy Cedric
I prepared some photo to explain to you the replacement steps of ccd. But before to change it i suggest to you to replace only the dragonfly board and the flat cable and try if the camera is fixed.
if the problem is solved, avoid to you a very delicate intervention (ccd replace)

cedricgoubard, 2021-09-14:
Hi @Uboldi8
Thank you for the advice, I'll change the card and cable as soon as I receive the package.
Speaking of which, could you give me a status update? Have you already sent it?

Uboldi80, 2021-09-14:
hy @cedricgoubard 

yes, the package was delivered, i try asking where it went.

pattacini, 2021-11-24:
Hi @Uboldi80 

Just realized that you have opened again this issue.
To prevent the automation from marking issues as `stale`, you could assign the label `pinned`.

Uboldi80, 2021-11-25:
thankyou @pattacini !

cedricgoubard, 2023-09-06:
Hi @sgiraz,

Great, let me know if I can help with anything.

cedricgoubard, 2023-10-19:
Hi @sgiraz,

Any news on this? How is the repair going?

sgiraz, 2023-10-19:
Hi @cedricgoubard,

### Good news ✅
- We started to work on the robot
- The DragonFly cameras seem working without any problems.

### Bad news ❌
- We had to replace the PC104 board because the one mounted on the robot was faulted.
- The robot presents many more issues than the one mentioned here and in #1205 (missing/wrong screws, damaged/missing covers, rust around the motors.

>**Note**
>The preliminary checks on the robot are still in progress, and we expect to find other mechanical and/or electrical issues.
>The repair cost estimate may need to be revised after the first inspection is complete.

cedricgoubard, 2023-10-24:
Hi @sgiraz,

Thank you for the update, nice to see you are making progress.
Could you please let us know before you start any work that exceeds the original quote so that we can see if / how we could budget for it?

sgiraz, 2023-10-27:
Hi @cedricgoubard,

As of now, the robot passed all the preliminary checks ✅:
- [x] Check if the [PC104 CPU board](http://wiki.icub.org/wiki/CPU_board_PC104) is working correctly (replaced from spare parts)
- [x] Check if the [CFW card](http://wiki.icub.org/wiki/CFW_card) is working correctly
- [x] Check if the Dragon Fly boards are working correctly
- [x] Check if the laptop is working correctly
- [x] Report all **macroscopic** broken mechanics and electronics parts onboard the robot.

Hence, at the moment, it is not necessary to revise the cost estimation for the maintenance. But the activity is still in progress.
In case we encounter a broken/faulted part that needs attention from the economic point of view, we'll advise you in time.

cedricgoubard, 2024-01-12:
Hi @sgiraz,
Happy new year!

Do you have any updates on our iCub?

cedricgoubard, 2024-01-15:
Hi @sgiraz,
Thank you for the update, we are glad to hear you are making progress 🥳 

Could you please share some more details on the extra time/cost it would take to fix the legs and the fingers?  We understand those would be estimates, but they would help us decide the best course of action.

sgiraz, 2024-01-29:
Hi @cedricgoubard,

I'm sorry for the late reply. Finally, We brought the robot back to life! 🤖

https://github.com/robotology/icub-tech-support/assets/18039045/14476b8e-9b15-4401-bc09-cde31e676a25

Here is the list of the joints currently working correctly ✅:
- torso: roll, pitch, yaw
- right/left shoulder: roll, pitch, yaw
- right/left elbow
- right/left wrist: pronosupination, pitch, yaw
- right/left hand:
  - thumb oppose (adduction)
  - thumb proximal
  - thumb distal
  - index proximal
  - index distal
  - middle proximal
  - middle distal
  - pinky
 - neck: roll, pitch, yaw
 - eyes: tilt, version, vergence
 

> Could you please share some more details on the extra time/cost it would take to fix the legs and the fingers? We understand those would be estimates, but they would help us decide the best course of action.

As mentioned before, We currently do not have any `BLL` electronics boards available to repair both legs. These boards are old (deprecated) and very difficult to obtain, so it would take some time to acquire them. Additionally, we have some high-priority demos that we need to support in the next month, so we may not be able to work on `iCubLondon01` continuously.
Based on these factors, we estimate that it would take about 2-3 months to repair the legs and the finger abduction joints (+ the shipping time).

>[!Note]
>Consider that if you do not need the use of these joints, we could begin the shipping process for the robot as early as next week.

Please let us know if you have any other questions.

cedricgoubard, 2024-01-29:
Hi @sgiraz,

That's great news, thank you! Feels great to see our iCub moving again 😄 

We do not mind giving up on the legs, but the robot's utility will be much less to us without all five fingers working. 
We would rather wait the extra 2 months and see if you can also repair those.

sgiraz, 2024-02-01:
Hi @cedricgoubard,

> We do not mind giving up on the legs, but the robot's utility will be much less to us without all five fingers working.

Just to avoid any misunderstandings, I want to clarify that the open-close function of all five fingers of both hands is working correctly, as shown in the video below:

https://github.com/robotology/icub-tech-support/assets/18039045/dc78c7ba-a188-43e4-80ce-0207794d4393

The joint that is currently not working is the finger adduction joint, which is responsible for the following movement:

https://github.com/robotology/icub-tech-support/assets/18039045/5a7cbb62-c04b-4f02-9244-2fe4f90d2f25

So, if you prefer to have the fingers adduction working as well, we can proceed to repair them by the next 2 months, otherwise we can prepare the robot for the shipment starting from the next week. 
Let us know what is better for you. 😊

cedricgoubard, 2024-02-05:
Hi @sgiraz, thank you for clarifying!
We would still rather get the fingers fully fixed, we do not mind waiting an extra 2 months 😄 

Let me know if there is anything we can do on our end.

AntonioConsilvio, 2024-04-11:
Hi @cedricgoubard, We have successfully restored the functioning of the adduction joints! 

https://github.com/robotology/icub-tech-support/assets/114915464/e8ce122c-332a-4d37-bf06-c364cf1f483c

https://github.com/robotology/icub-tech-support/assets/114915464/dc72b32b-f664-433d-83ed-872baa112f66

Here is the list of the joints currently working correctly ✅:

- torso: roll, pitch, yaw
- right/left shoulder: roll, pitch, yaw
- right/left elbow
- right/left wrist: pronosupination, pitch, yaw
- right/left hand:
  - fingers adduction
  - thumb oppose (adduction)
  - thumb proximal
  - thumb distal
  - index proximal
  - index distal
  - middle proximal
  - middle distal
  - pinky
 - neck: roll, pitch, yaw
 - eyes: tilt, version, vergence

We have also successfully tested ✅:

- face expression
- cameras
- skins

Since the robot works properly (according to the agreed compromises) we have started the shipping process!

cc @Fabrizio69 @andreaderito

andreaderito, 2024-05-14:
Hi @cedricgoubard I've been informed that you should have received back the robot at your labs on 09.05.2024 - so, in case of questions just let us know. Best regards, Andrea
cc @AntonioConsilvio @Fabrizio69

AntonioConsilvio, 2024-05-17:
Hi @cedricgoubard! Since the maintenance service has been done and the robot has been shipped back, I will proceed with closing the issue, if there are any other problems feel free to open a new one!

closed 2024-05-17 by AntonioConsilvio