robotology/mesh-support #1174: ssh key different then expected warning after icub-head update

traversaro, 2021-08-05:
### Device name 🤖

iCubGenova09 

### Request/Failure description

I recently updated iCubGenova09 icub-head to the `iCubOS_V8.0.20.04_20210604` image. After that, I followed the instructions in https://icub-tech-iit.github.io/documentation/icub_operating_systems/other-machines/icub-server-laptop/#install-the-ssh-keys-for-password-less-login-on-icub-head to ensure that the `icub-console` was able to ssh in icub-head without any password prompt.

However, now whenever I ssh in icub-head, I get this warning:
~~~
icub@iiticublap158:~ $ ssh icub-head
Warning: the ECDSA host key for 'icub-head' differs from the key for the IP address '10.0.0.2'
Offending key for IP in /home/icub/.ssh/known_hosts:1
Matching host key in /home/icub/.ssh/known_hosts:4
Are you sure you want to continue connecting (yes/no)? yes
Welcome to iCub OS version 8.0.20.04 rel. 20210604
Based on Ubuntu 20.04.2 LTS (GNU/Linux 5.4.0-80-lowlatency x86_64)
~~~

I guess it is related in some way to the old ssh key being used, but I wonder what is the best way of deal with this.

### Detailed context

See failure description.

traversaro, 2021-08-05:
@Uboldi80 as for https://github.com/robotology/icub-tech-support/issues/1172#issuecomment-892828731, as this is mostly a IT issue I directly assigned @mbrunettini and @davidelasagna .

davidelasagna, 2021-08-05:
Yes this issue was related to the old ssh key used fixed  by removing the keys with  `ssh-keygen -R 10.0.0.2` this issue could and would probably pop up also in the other machines of the setup that used to ssh into the iCub 3 head

traversaro, 2021-08-05:
Thanks, I imagined this, but I did not know about the command:
~~~
ssh-keygen -R 10.0.0.2
~~~

closed 2021-08-05 by traversaro