robotology/mesh-support #1175: Robot Motors does not start anymore

GiulioRomualdi, 2021-08-05:
### Device name 🤖

iCubGenova04

### Request/Failure description

While testing the current controller on the pole the robot went in over-current.

After this event, the robot motors do not start anymore and the motor button led blinks.

cc @Uboldi80 @Giulero  

### Detailed context

The head starts as usual.

Here the log
[log_icub-head_yarprobotinterface_1669.txt](https://github.com/robotology/icub-tech-support/files/6938592/log_icub-head_yarprobotinterface_1669.txt)

S-Dafarra, 2021-08-06:
cc @DanielePucci @maggia80

maggia80, 2021-08-06:
@GiulioRomualdi could you run the current controller for each part of the robot separately, to better understand if there is a particular part that causes the over-current of the whole robot.

S-Dafarra, 2021-08-06:
> @GiulioRomualdi could you run the current controller for each part of the robot separately, to better understand if there is a particular part that causes the over-current of the whole robot.

I think that the problem is that the motors do not turn on at all anymore

maggia80, 2021-08-06:
ah ok! so we need to figure out which board is causing the short-circuit.

GiulioRomualdi, 2021-08-06:
When we tried to switch on the motors none of the boards blinks

maggia80, 2021-08-11:
board 3B6 was in short circuit:
[image]

we replaced ( thanks @emilianob80 ) the 3B6 with a new one.
The problem was caused by a screw:

[image]

[image]

you can now use the robot even if the covers of the left upper leg are not mounted.

S-Dafarra, 2021-08-11:
Oh wow, a screw! Super, thanks a lot!

DanielePucci, 2021-08-11:
Big thanks @maggia80 and @emilianob80 for the support, having the robot functional this week was an important milestone for performing tests of a looming outdoor demo in September. Grazie ragazzi

DanielePucci, 2021-08-11:
Big thanks @maggia80 and @emilianob80 for the support, having the robot functional this week was an important milestone for performing tests of a looming outdoor demo in September. Grazie ragazzi!

P.s. I belive that this is indeed another lesson learnt. We may think of designing the next generation of the robot electronics so as to avoid this kind of occurrence (e.g. encapsulate the electronics, or having the robot simply fully covered by covers)

GiulioRomualdi, 2021-08-11:
@maggia80  and @emilianob80, thank you so much!

S-Dafarra, 2021-08-12:
Thanks a lot to @maggia80 which connected the leg back. I have tested it and the robot seems ok. Is it possible to put the covers back on?

DanielePucci, 2021-08-17:
CC @Fabrizio69

emilianob80, 2021-08-18:
> Thanks a lot to @maggia80 which connected the leg back. I have tested it and the robot seems ok. Is it possible to put the covers back on?

@S-Dafarra Of course, if the robot is working the covers can be reassembled

S-Dafarra, 2021-08-18:
> > Thanks a lot to @maggia80 which connected the leg back. I have tested it and the robot seems ok. Is it possible to put the covers back on?
> 
> @S-Dafarra Of course, if the robot is working the covers can be reassembled

Yes, please!

closed 2021-08-23 by Uboldi80