robotology/mesh-support #1177: iCub version & repair tips

cedricgoubard, 2021-08-09:
### Device name 🤖

iCubLondon01

### Request/Failure description

How can we know our iCub's version, and where can we find resources for fixing cables in the arms?

### Detailed context

Hello, 

We are planning on fixing a few issues in our iCub (unresponsive index finger, broken cables in wrist...).
I found the relevant documentation, but I am not sure of our robot's version... Given [the descriptions](https://icub-tech-iit.github.io/documentation/icub_versions/), I would assume 1.4 or 2.0; how could I make sure?

Also, do you have any other advice/tutorials before we start?

Thank you for your help.

maggia80, 2021-08-10:
Hi @cedricgoubard,

you can find the version of your robot here: https://robot-bazaar.iit.it/robots/collection/robots/iCubLondon01

For the lower arm you should follow:

https://icub-tech-iit.github.io/documentation/icub_tendons/remove/#1-lower-arm-v1

and then:

https://icub-tech-iit.github.io/documentation/icub_tendons/lower_arm_V1/

take pictures before unplugging connectors, it's quite old your robot and there could be some differences wrt to documentation.

cedricgoubard, 2021-08-11:
Great, thank you @maggia80!
We fixed the broken motors in the hand, but now the elbow motor encoder is broken... We are trying to solder it back, but in case it doesn't work, is there any way we could get a new one (preferably with the wire already soldered)?

This is the encoder from the wiring documentation:
[image]

cedricgoubard, 2021-08-12:
Hi @maggia80,
We managed to solder it, let's hope it does not break again :smile: 
Thanks again for your help!

closed 2021-08-12 by cedricgoubard