robotology/mesh-support #1178: Firmware version error in yarprobotinterface

cedricgoubard, 2021-08-12:
### Device name 🤖

iCubLondon01

### Request/Failure description

yarprobotinterface will not start beacause of a version difference with the iCub firmware. We apparently need a `network.iCubLondon01.xml` to perform the update, but I cannot find it in [here](https://github.com/robotology/robots-configuration/tree/master/iCubLondon01).

### Detailed context

After fixing a broken tendon, I tried to recalibrate our iCub, but yarprobotinterface seems to have some issues with the iCub firmware.

<details> 
<summary>Here is the whole log trace.</summary>

```bash
icub@pc104:~$ yarprobotinterface 
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/build-pc104/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /usr/local/src/robot/robotology-superbuild/build-pc104/install/share/ICUBcontrib/robots/iCubLondon01/./icub_ikart.xml
[DEBUG] Preprocessor complete in:  0.490711 s
[INFO] |yarp.os.Port| Port /icub/yarprobotinterface active at tcp://10.0.2.2:10008/
[INFO] startup phase starting...
[INFO] ***** Configuring cartesian controller *****
[INFO] Acquiring options for group GENERAL...
[INFO] Acquiring options for group DRIVER_0...
[INFO] Acquiring options for group DRIVER_1...
[INFO] PLANT_MODEL group detected
[INFO] DH Table:  (H0 (0.0 -1.0 0.0 0.0 0.0 0.0 -1.0 0.0 1.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0)) (HN (1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0)) (link_0 (A 0.0320000000000000006661) (D 0.0) (alpha 90.0) (offset 0.0) (min -22.0) (max 84.0) (blocked 0.0)) (link_1 (A 0.0) (D -0.00549999999999999968081) (alpha 90.0) (offset -90.0) (min -39.0) (max 39.0) (blocked 0.0)) (link_2 (A 0.0233646999999999986641) (D -0.143300000000000010703) (alpha -90.0) (offset 105.000000000000014211) (min -59.0000000000000071054) (max 59.0000000000000071054) (blocked 0.0)) (link_3 (A 0.0) (D 0.107740000000000002434) (alpha -90.0) (offset 90.0) (min -95.5) (max 5.0)) (link_4 (A 0.0) (D 0.0) (alpha 90.0) (offset -90.0) (min 0.0) (max 160.800000000000011369)) (link_5 (A 0.0149999999999999994449) (D 0.152279999999999998694) (alpha -90.0) (offset 75.0) (min -37.0) (max 100.0)) (link_6 (A -0.0149999999999999994449) (D 0.0) (alpha 90.0) (offset 0.0) (min 5.5) (max 106.0)) (link_7 (A 0.0) (D 0.137300000000000005373) (alpha 90.0) (offset -90.0) (min -50.0) (max 50.0)) (link_8 (A 0.0) (D 0.0) (alpha 90.0) (offset 90.0) (min -65.0) (max 10.0)) (link_9 (A 0.0625) (D -0.0160000000000000003331) (alpha 0.0) (offset 0.0) (min -25.0) (max 25.0)) (numLinks 10) (type left)
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/in active at tcp://10.0.2.2:10009/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/out active at tcp://10.0.2.2:10010/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/cartesianSolver/left_arm/rpc active at tcp://10.0.2.2:10011/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/command:i active at tcp://10.0.2.2:10012/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/state:o active at tcp://10.0.2.2:10013/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/events:o active at tcp://10.0.2.2:10014/
[INFO] |yarp.os.Port| Port /icub/cartesianController/left_arm/rpc:i active at tcp://10.0.2.2:10015/
[INFO] |yarp.dev.PolyDriver| Created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
[INFO] ***** Configuring cartesian controller *****
[INFO] Acquiring options for group GENERAL...
[INFO] Acquiring options for group DRIVER_0...
[INFO] Acquiring options for group DRIVER_1...
[INFO] PLANT_MODEL group detected
[INFO] DH Table:  (H0 (0.0 -1.0 0.0 0.0 0.0 0.0 -1.0 0.0 1.0 0.0 0.0 0.0 0.0 0.0 0.0 1.0)) (HN (1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0 0.0 0.0 0.0 0.0 1.0)) (link_0 (A 0.0320000000000000006661) (D 0.0) (alpha 90.0) (offset 0.0) (min -22.0) (max 84.0) (blocked 0.0)) (link_1 (A 0.0) (D -0.00549999999999999968081) (alpha 90.0) (offset -90.0) (min -39.0) (max 39.0) (blocked 0.0)) (link_2 (A -0.0233646999999999986641) (D -0.143300000000000010703) (alpha 90.0) (offset -105.000000000000014211) (min -59.0000000000000071054) (max 59.0000000000000071054) (blocked 0.0)) (link_3 (A 0.0) (D -0.107740000000000002434) (alpha 90.0) (offset -90.0) (min -95.5) (max 5.0)) (link_4 (A 0.0) (D 0.0) (alpha -90.0) (offset -90.0) (min 0.0) (max 160.800000000000011369)) (link_5 (A -0.0149999999999999994449) (D -0.152279999999999998694) (alpha -90.0) (offset -105.000000000000014211) (min -37.0) (max 100.0)) (link_6 (A 0.0149999999999999994449) (D 0.0) (alpha 90.0) (offset 0.0) (min 5.5) (max 106.0)) (link_7 (A 0.0) (D -0.137300000000000005373) (alpha 90.0) (offset -90.0) (min -50.0) (max 50.0)) (link_8 (A 0.0) (D 0.0) (alpha 90.0) (offset 90.0) (min -65.0) (max 10.0)) (link_9 (A 0.0625) (D 0.0160000000000000003331) (alpha 0.0) (offset 180.0) (min -25.0) (max 25.0)) (numLinks 10) (type right)
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/in active at tcp://10.0.2.2:10016/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/out active at tcp://10.0.2.2:10017/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/cartesianSolver/right_arm/rpc active at tcp://10.0.2.2:10018/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/command:i active at tcp://10.0.2.2:10019/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/state:o active at tcp://10.0.2.2:10020/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/events:o active at tcp://10.0.2.2:10021/
[INFO] |yarp.os.Port| Port /icub/cartesianController/right_arm/rpc:i active at tcp://10.0.2.2:10022/
[INFO] |yarp.dev.PolyDriver| Created device <cartesiancontrollerserver>. See C++ class ServerCartesianController for documentation.
[INFO] |yarp.devices.ControlBoardWrapper| /icub/left_arm : no ROS initialization required
[INFO] |yarp.devices.ControlBoardWrapper| /icub/left_arm  initting YARP initialization
[INFO] |yarp.os.Port| Port /icub/left_arm/rpc:i active at tcp://10.0.2.2:10023/
[INFO] |yarp.os.Port| Port /icub/left_arm/command:i active at tcp://10.0.2.2:10024/
[INFO] |yarp.os.Port| Port /icub/left_arm/state:o active at tcp://10.0.2.2:10025/
[INFO] |yarp.os.Port| Port /icub/left_arm/stateExt:o active at tcp://10.0.2.2:10026/
[INFO] |yarp.dev.PolyDriver| Created wrapper <controlboardwrapper2>. See C++ class ControlBoardWrapper for documentation.
[INFO] |yarp.devices.ControlBoardWrapper| /icub/right_arm : no ROS initialization required
[INFO] |yarp.devices.ControlBoardWrapper| /icub/right_arm  initting YARP initialization
[INFO] |yarp.os.Port| Port /icub/right_arm/rpc:i active at tcp://10.0.2.2:10027/
[INFO] |yarp.os.Port| Port /icub/right_arm/command:i active at tcp://10.0.2.2:10028/
[INFO] |yarp.os.Port| Port /icub/right_arm/state:o active at tcp://10.0.2.2:10029/
[INFO] |yarp.os.Port| Port /icub/right_arm/stateExt:o active at tcp://10.0.2.2:10030/
[INFO] |yarp.dev.PolyDriver| Created wrapper <controlboardwrapper2>. See C++ class ControlBoardWrapper for documentation.
[INFO] |yarp.devices.ControlBoardWrapper| /icub/head : no ROS initialization required
[INFO] |yarp.devices.ControlBoardWrapper| /icub/head  initting YARP initialization
[INFO] |yarp.os.Port| Port /icub/head/rpc:i active at tcp://10.0.2.2:10031/
[INFO] |yarp.os.Port| Port /icub/head/command:i active at tcp://10.0.2.2:10032/
[INFO] |yarp.os.Port| Port /icub/head/state:o active at tcp://10.0.2.2:10033/
[... 535 lines cut ...]
[ERROR] |yarp.os.Time|  - Call useSystemClock()
[ERROR] |yarp.os.Time|  otherwise use yarp::os::SystemClock::nowSystem() and yarp::os::SystemClock::delaySystem() instead of Time::now() and Time::delay()
```

</details>

As you can see, the first error message is:
```bash
[ERROR] 
[ERROR] ###################################################################################
[ERROR] ###################################################################################
[ERROR] 
[ERROR]   It has been detected that your control boards are not using the same
[ERROR]   CAN protocol used by yarprobotinterface. yarprobotinterface cannot continue.
[ERROR]   Please update your system (yarprobotinterface and/or your control board firmware.
[ERROR]   For further information please visit: https://icub-tech-iit.github.io/documentation/icub_firmware
[ERROR] 
[... cut ...]
[ERROR] checkFirmwareVersions() failed. CanBusMotionControl::open returning false,
[ERROR] |yarp.dev.PolyDriver| Driver <canmotioncontrol> was found but could not open
```

I then tried to use the firmware update script `scripts/FirmwareUpdater.script.verify.all.sh` as described in [the documentation](https://icub-tech-iit.github.io/documentation/icub_firmware/firmware/firmware/), but it requires a `network.iCubLondon01.xml` file which I do not have, and the documentation said to ask for it here.

pattacini, 2021-08-12:
Hi @cedricgoubard 

It's mid-August, so if you could wait until next week, we will be happy to help you with this.

davidetome, 2021-08-16:
Hi @cedricgoubard ,

 could you please open `FirmwareUpdater` , make a `Discover` on each `CFW2 "X"` node and then drop the result here?

cedricgoubard, 2021-08-16:
Hi @davidetome,
Thank you for coming back to me. Here is the result:
[image]

davidetome, 2021-08-16:
Hi @cedricgoubard ,

 It looks like you're missing few boards. In particular, referring to the [doc](https://icub-tech-iit.github.io/documentation/icub_firmware/associated-firmware/associated-firmware/#icub-v130), we miss: 

## CFW2 <3> - Left Leg
| Board  |     CAN ID |      
|----------|:-------------:|
| bll |    5|
| bll |    6|
| bll |    7|
| strain | 13|

## CFW2 <6> - Right Hand
| Board  |     CAN ID |      
|----------|:-------------:|
| mc4 |  5|
| mais | 14|

## CFW2 <8> - Skin Right hand
| Board  |     CAN ID |      
|----------|:-------------:|
| mtb |  8|

Before investigating the reason you could try to power off/on the boards and try to `Discover` again such missing boards.
The boards FW [revisions](https://github.com/robotology/icub-firmware-build/blob/master/info/firmware.info.xml) are up-to-date a part of `mtb` and `strain` boards (we can also update them using `FirmwareUpdater`). 
Anyway, first of all, we need to have all boards available and, In the meanwhile, I can prepare the `network.iCubLondon01.xml `.

cedricgoubard, 2021-08-16:
Hi @davidetome,
Thank you for clearing this up. The skin right-hand board is no longer missing (it just was because I unplugged it to fix a cable). 

For the right-hand boards, can you confirm that these boards are located in the biceps? If so, I might have plugged them back wrong when I fixed the arm, I can take a look.

As for the left leg, I never opened it... Are the boards located in the thigh? Should I follow [this guide](https://icub-tech-iit.github.io/documentation/icub_tendons/waist_legs/) to check?

davidetome, 2021-08-17:
> For the right-hand boards, can you confirm that these boards are located in the biceps?

the `mc4` is in the bicep, the `mais` board is on the hand

> As for the left leg, I never opened it... Are the boards located in the thigh? Should I follow this guide to check?

the boards are placed as described [here](https://github.com/icub-tech-iit/electronics-wiring/blob/master/icub1/iCubWiring_Ver.1.3.0/PDF/Leftleg_V1.3.0/leftleg_wiring_figures_V1.3.0.pdf); you should check if they are powered and if the CAN line 3 is not cut

cedricgoubard, 2021-08-18:
Hi @davidetome,
I fixed the right-hand boards:
[image]

However, I cannot find what is wrong with the leg. All the boards are lighting up:
[image]

[image]

That being said, we do not plan on using the leg, so I don't mind leaving it like that if you think it is possible.

davidetome, 2021-08-18:
Ciao @cedricgoubard ,

> That being said, we do not plan on using the leg, so I don't mind leaving it like that if you think it is possible.

Since the boards are powered, probably there's a problem in the CAN line in terms of wiring or, worst case, the relative CW2 CAN channel is broken.
Anyway, you can go on since in your `icub_ikart.xml` [file](https://github.com/robotology/robots-configuration/blob/master/iCubLondon01/icub_ikart.xml) both legs are not included 👍🏻

cedricgoubard, 2021-08-18:
Hi @davidetome,

Thank you for your help; let me know if you need any information on my part for the `network.iCubLondon01.xml` file.

Also, an unrelated question: could tell me what you are using to glue the wires to a connector after crimping/soldering? It looks really useful to prevent the wires from breaking again.
[image]

davidetome, 2021-08-20:
Ciao @cedricgoubard ,

> Thank you for your help; let me know if you need any information on my part for the network.iCubLondon01.xml file.
Here is the `network.iCubLondon01.xml` file : 
```xml
<?xml version="1.0"?>

<!--    description 

        this file contains the boards mounted on the robot grouped by body parts.

        there is only one group called host which contains name and as many rows as the supported
        devices. each one is tagged <hasdevice> and contains a type (ETH or CFW) and an address 
        (in ip format or just a number for the case of CFW).
        
        then there are many groups called part. each one may contain many groups called boards: 
        one per physical board on the robot on that part.
        
        every <board> has two compulsory properties: type, name. it may also have an 
        additional property: required. so far the only value of required is "version".
        each <board> contains three compulsory fields: ondevice, ataddress, connected. 
        and it may have an additional field: version (if there is the required="version" property).
        here is description of each of the previous:
        -   type: must have a valid board name in short format: ems4, mc4plus, foc, strain, 
            strain2, mtb, mtb4, mais, mc4, etc.
        -   name: can be any string.
        -   required: if present can be only ="version". it tells that that board MUST match
            a given version of firmware. it is used to identify amongst the many versions of mc4 
            boards.
        -   ondevice: tells on which device the board is located. it can be ETH (for ETH robots)
            or CFW for first generation CAN robots. 
        -   ataddress: contains the network address under device. if device is ETH there is
            ip="10.0.1.x" and if the board is a CAN board there is also canbus="c" canadr="a".
            if the device is CFW, then there is only canbus="c" canadr="a".
        -   connected: contains the connection information. if bus ETH there are: prev and next
            which are the previous and the next ip addresses (they both must be present). the last
            board must have next="none". if bus is CAN there are no extra fields.
        -   version: contains the required version. typically only major and minor values are specified.
         
        the aim of this files is to be parsed by the python program robotFWmanager.py
  -->

<robotnetwork>

    <host name="pc104" >
        <hasdevice type= "CFW" address="0" />
        <hasdevice type= "CFW" address="1" />
        <hasdevice type= "CFW" address="2" />
        <hasdevice type= "CFW" address="3" />
        <hasdevice type= "CFW" address="4" />
        <hasdevice type= "CFW" address="5" />
        <hasdevice type= "CFW" address="6" />
        <hasdevice type= "CFW" address="7" />
        <hasdevice type= "CFW" address="8" />
        <hasdevice type= "CFW" address="9" />
    </host>

    <part name="head">

        <board type='bll' name="bll.head.CAN0:1" required="version">
            <version major="1" minor="98" />
            <ondevice>CFW</ondevice>
            <ataddress canbus="0" canadr="1"  />
            <connected bus="CAN" />
        </board>
        
        <board type='mc4' name="mc4.head.CAN0:2" required="version">
            <version major="2" minor="21" />
            <ondevice>CFW</ondevice>
            <ataddress canbus="0" canadr="2" />
[... 327 lines cut ...]
            <connected bus="CAN" />
        </board>

        <board type='mtb' name="mtb.right_leg.foot.CAN9:14">
            <ondevice>CFW</ondevice>
            <ataddress canbus="9" canadr="14" />
            <connected bus="CAN" />
        </board>

    </part>

    
    <part name="custom"> 
    </part>

    <part name="test">
    </part>

</robotnetwork>
```
All control boards FW [revisions](https://github.com/robotology/icub-firmware-build/blob/master/info/firmware.info.xml) are up-to-date a part of `strain` and `mtb`

> Also, an unrelated question: could tell me what you are using to glue the wires to a connector after crimping/soldering? It looks really useful to prevent the wires from breaking again.

I think it is `Pattex` glue, but I'm going to ask the guys who work in the iCub wiring for the `RS` code and let you know.

cedricgoubard, 2021-08-21:
Hi @davidetome,

Thank you for the file, the firmware update worked!
I will now move on to calibrating the robot; I found [this tutorial](https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration/) for robots 2.X, but ours is a 1.X... Can it work for us too?

Also, I noticed that most of our files use a calibration type 3, which is not referenced [here](https://icub-tech-iit.github.io/documentation/icub_r1_icub3_calibration_types/icub_r1_calibration_types/). Should I use type 12 instead, which is described as the  "newest version of calibration 3"?

davidetome, 2021-08-23:
Ciao @cedricgoubard ,

 regarding the [glue](https://github.com/robotology/icub-tech-support/issues/1178#issuecomment-900988618), my colleagues confirmed they're using `Pattex`;  no more available on `RS` but I link you the amazon [page](https://github.com/robotology/icub-tech-support/issues/1178#issuecomment-900988618)

>  I found this tutorial for robots 2.X, but ours is a 1.X... Can it work for us too?

No, it's intended for 2.5> ETH based iCub robots

> Should I use type 12 instead, which is described as the "newest version of calibration 3"?

Unfortunately no, type 12 works only on the newest ETH boards.
You have to follow this [peocedure](https://github.com/robotology/icub-tech-support/files/7030963/TP_ROBOT_CALIBRATION.pdf) using the `.xls` file in your robot's folder, taking care that it is outdated in terms of SW used, in particular : 

- `robotInterface` is know `yarprobotinterface`
- `motorGui` is know `yarpmotorgui`

Anyway we've planned to port/update the old procedure on the official doc site

cedricgoubard, 2021-08-24:
Hi @davidetome,
Thank you for pointing out the calibration resources! I'll close this for now since the firmware update is done, and I'll start working on the calibration.

closed 2021-08-24 by cedricgoubard