robotology/mesh-support #1179: Right hand often fails to calibrate and right thumb oscillate

S-Dafarra, 2021-08-23:
### Device name 🤖

iCubGenova04

### Request/Failure description

The right hand often fails to calibrate, and the thumb, the index and the middle fingers remain closed.

### Detailed context

The right hand fails to calibrate quite often. Even when it calibrates, the right thumb keeps oscillating. We also noticed several MAIS timeouts when starting it up. I guess that the main reason might be the thumb itself, which does not move properly at startup.

S-Dafarra, 2021-08-23:
cc @DanielePucci

Uboldi80, 2021-08-23:
Hy @S-Dafarra 

Maybe is better to remind this problem in the issue 1139.
Can  you close this issue please? 
perhaps it would be better to describe this problem in the issue which already describes the swing of the right thumb and the attempts to repair it.

S-Dafarra, 2021-08-23:
Hi @Uboldi80. Yes, you are right, this is a duplicate. Closing this then

closed 2021-08-23 by S-Dafarra