robotology/mesh-support #1185: iCubGenova04 - replace the power connector

vpunithreddy, 2021-08-30:
### Device name 🤖

iCubGenova04

### Request/Failure description

@gabrielenava  noted that the power connector of iCubGenova04 is very damaged:

[image]

It would be nice to replace it.

### Detailed context

The knurling for the connector is damaged and is difficult to assemble and dismantle the connector for the power cable.
The connection is not secure since the robot is flying. It would be better to have a more secure connection by replacing the connector.

Gandoo, 2021-09-01:
[image]

[image]

un pin dove e saldato il filo è spezzato, la causa sicuramente dovuta alla rotazione della curva durante l'inserimento del connettore. **Non bisogna usale una pinza ma usare solo la mano per avvitare la ghiera**

cc  @DanielePucci @FabioBergonti @maggia80 @Uboldi80 @Fabrizio69 @gabrielenava

Uboldi80, 2021-09-01:
replacement of power connector done.

Thankyou @Gandoo

FabioBergonti, 2021-09-01:
Thank you @Gandoo 

>un pin dove e saldato il filo è spezzato, la causa sicuramente dovuta alla rotazione della curva durante l'inserimento del connettore. 
**Non bisogna usale una pinza ma usare solo la mano per avvitare la ghiera**

@robotology/iit-dynamic-interaction-control probably most of you already know, but I remind you that the black nut must be tightened (by hand), while the grey one must not be touched/rotated

[image]

closed 2021-09-01 by Uboldi80

DanielePucci, 2021-09-03:
@robotology/iit-artificial-mechanical-intelligence indeed an important comment by @FabioBergonti , i.e.
> probably most of you already know, but I remind you that the black nut must be tightened (by hand), while the grey one must not be touched/rotated