robotology/mesh-support #1186: R1SN003 - left arm broken

lnobile, 2021-09-02:
### Device name 🤖

R1SN003

### Request/Failure description

Left arm JOINT 4 (l_wrist_pronosupination_joint) doesn't work properly.

### Detailed context

The motor and encoder are working correctly but even if I can hear the motor spinning the joint doesn't move. Probably a loosen/broken belt.

maggia80, 2021-09-07:
@fbiggi @emilianob80 could you check it?

emilianob80, 2021-09-07:
@maggia80 Ok I do a first cechk myself because @fbiggi is in the mechanical workshop these days. If you are unable to solve it independently, we schedule an intervention with @fbiggi 

cc: @Uboldi80

emilianob80, 2021-09-08:
@lnobile The screw indicated in the red circle in the following image was loose. The problem was solved and the rotation with the jogger tested

[image]

cc: @maggia80 @fbiggi

closed 2021-09-08 by emilianob80