robotology/mesh-support #1190: iCubGenova04 broken tendon right thumb

isorrentino, 2021-09-17:
### Device name 🤖

iCubGenova04

### Request/Failure description

Broken right thumb tendon.

### Detailed context

Right thumb tendon broke as soon as I started the yarprobotinterface, in particular while the hand calibration.

cc @Uboldi80

Uboldi80, 2021-09-20:
the problem of the broken tendon was probably caused by the gearbox which turned very badly.
this in fact involved the subsequent tearing of another tendon of the same joint.
furthermore, the encoder reading was incorrect and the joint always started in NOT CALIBRATED STATUS, even though this joint had been calibrated at least 3 more times.
the problem was due to the ground of the connector which was uncovered and was going to touch another cable of the connector altering the values continuously.

S-Dafarra, 2021-09-20:
I guess there are still issues with the encoders. At startup, this time the oppose joint went to the other end a couple of times, reading around 140deg. After a few startup, another tendon got broken

[image]

closed 2021-09-27 by Uboldi80