robotology/mesh-support #1192: temporary increase to eye tilt motor

lunagava, 2021-09-20:
### Device name 🤖

iCub[Genova][02]

### Request/Failure description

the robot cannot maintain a steady eye tilt, and cannot control to a desired position

### Detailed context

Due to the custom eye-bulb in Genova02 we previously, we have given more power to eye tilt joint. It seems in the robot-configurations file, the values have been reset to the original value (a re-install of the icub-head was recently performed). I think we need to tweak the power given to the tilt motor again. We would like some help with this as we aren't sure exactly which values in the robots-configuration should be changed to achieve this.

pattacini, 2021-09-20:
Hi @lunagava 

Can you be more precise on this? For example, by pointing to the values in the XML files you're referring to?

Anyway, in our analysis of the problem, we've recently come up with the hypothesis that the motor shaft may slip being sometimes unable to deliver the torque to the joint (see the figure below).

[image]

Thus, we've planned to check this out during our current sprint (weeks 20/09 - 01/10)

arrenglover, 2021-09-20:
I think it was this file: 
https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/hardware/motorControl/head-eb21-j2_5-mc.xml

From memory we had a bit of a boost to the motor to make things work temporarily. Now something has really broken again and we can barely keep the eyes controlled - so I was thinking this change may have been reset by some recent re-install. I don't know enough about the exact parameters to tweak.

We were hoping to give a demo tomorrow so if someone is around tomorrow morning that can check either the file and the shaft/bolt it would be appreciated.

pattacini, 2021-09-20:
Hi @arrenglover 

We cannot help with the motor shaft slippage as this activity needs to be discussed and planned carefully.

The motor parameters you could tweak are:
- `motorOverloadCurrents`
- `motorNominalCurrents`
- `motorPeakCurrents`

At any rate, they're already quite higher than the standard values. Compare these quantities between iCubGenova09 and iCubGenova02, for example:
- https://github.com/robotology/robots-configuration/blob/devel/iCubGenova09/hardware/motorControl/head-eb21-j2_5-mc.xml
- https://github.com/robotology/robots-configuration/blob/master/iCubGenova02/hardware/motorControl/head-eb21-j2_5-mc.xml

You could check whether the XML files actually used by your robot are the same in the upstream repo.

pattacini, 2021-09-20:
Anyway, when @davidetome carried out some acquisitions on iCubGenova09 (August 06) he got the following results: https://github.com/icub-tech-iit/study-improve-head-edpr/issues/5#issuecomment-894188615.

As you can see from the video, the problem was there without doing anything special with the XML params.

@davidetome, please correct me if I'm wrong.

arrenglover, 2021-09-21:
@pattacini thanks for the follow-up. 

The behaviour that I see in the @davidetome video is the long-standing behaviour we have had. The eyes are useable until trying to look down to the motor limits. however, we can still do our demo if we limit the eye motion to not look down so much, and don't move too fast.

The new behaviour is that the eye tilt cannot be held steady in nearly any position, even looking straight forward the back of the bulb drops, making the eyes look up.

I'm not 100% sure about the .xml file but have a vague memory that it was a temporary fix we did sometime since being at CRIS. I may be wrong. If anyone is available to check out the behaviour and see if there is a quick fix this morning it would be great, if not we go forward with the demo without robot motion. However, at the moment the robot is unusable for most of our tasks. Thanks again.

pattacini, 2021-09-21:
>I'm not 100% sure about the .xml file but have a vague memory that it was a temporary fix we did sometime since being at CRIS. I may be wrong.

Aside from the modifications I pointed out above in the XML files, I personally have no other suggestions: @ale-git do you have any?
No one from FIX team changed the local XML files; maybe, you could ask around.

I suspect that the slippage might have increased, but this morning our robot maintainers have been busy with iCubGenova04, unfortunately.

Anyway, I'm in touch w/ @Fabrizio69 to proceed within this week with the inspection of the motor.

Stay tuned.

ale-git, 2021-09-21:
@pattacini as you said the output limits are already at their maximum. Maybe increasing Kp and/or Ki?

pattacini, 2021-09-21:
>Maybe increasing Kp and/or Ki?

I'm afraid that this won't help as the PID already saturates (see the graph above).

pattacini, 2021-09-21:
Quick update on this: we're trying to organize the check tomorrow morning.

lunagava, 2021-09-21:
Thanks to the help of Jack, we opened the head to check the motor shaft, and by touching it, the eyes tilt started to work again properly, as you can see in the video. 

https://user-images.githubusercontent.com/45895918/134177258-47b8f3ea-edf0-4bec-84fa-d1d3bd38af2b.MOV

maggia80, 2021-09-21:
thanks @GiacomoVassallo13

arrenglover, 2021-09-22:
thanks @pattacini and @GiacomoVassallo13!

arrenglover, 2021-09-22:
I'll close this issue about the temporary problems and leave any further discussions on the long-term problem.

closed 2021-09-22 by arrenglover

pattacini, 2021-09-22:
Hi @arrenglover @lunagava 

Just a quick report of what we did this morning as for long-term discussion we do have an internal ticket.

With @GiacomoVassallo13 we checked that the bolt is screwed on the flat part of the shaft by moving the motor in different conditions while verifying the functionality

With @davidetome and @triccyx we acquired the data (see below) demonstrating that everything is working properly.

| In position | In PWM |
| :---: | :---: |
| [image] | [image] |

Therefore, we can basically conclude that our initial hypothesis has been verified successfully.
The tilt movement was flawed because of the loose bolt. The system anchoring the motor shaft to the joint has been redesigned to improve the robustness. We're testing it on one specific head.